prusa3d/PrusaSlicer #11939: Segementation fault when slicing

carrant, 2023-12-18:
### Description of the bug

OS: 
`cat /etc/os-release 
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy`

Crash when slicing stl file.

Unable to locate core file

### Project file & How to reproduce

Steps.
1. ./PrusaSlicer-2.7.1+linux-x64-GTK3-202312121425.AppImage
2. Drag Tyrannosaurus Rex Updated.stl to plate
3. Click "Slice" button
4. Crash
[crash.zip](https://github.com/prusa3d/PrusaSlicer/files/13709133/crash.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.1+linux-x64-GTK3-202312121425.AppImage

### Operating system

Ubuntu 22.04

### Printer model

MK4

lukasmatena, 2023-12-21:
I am not able to reproduce it. Can you please collect a backtrace using gdb? You can follow instructions in https://github.com/prusa3d/PrusaSlicer/issues/8168#issuecomment-1092799924 if you don't know how to do it.

carrant, 2023-12-21:
`Thread 68 "slic3r_tbb_10" received signal SIGFPE, Arithmetic exception.
[Switching to Thread 0x7fff58bfc640 (LWP 36155)]
0x00007ffff6b8712c in __atan_fma (x=0.087266460061073303) at ../sysdeps/ieee754/dbl-64/s_atan.c:97
97    ../sysdeps/ieee754/dbl-64/s_atan.c: No such file or directory.
(gdb) bt
#0  0x00007ffff6b8712c in __atan_fma (x=0.087266460061073303)
    at ../sysdeps/ieee754/dbl-64/s_atan.c:97
#1  0x000000000106992b in Slic3r::Arachne::VoronoiUtils::discretizeParabola(Slic3r::Point const&, Slic3r::Arachne::PolygonsSegmentIndex const&, Slic3r::Point, Slic3r::Point, int, float) ()
#2  0x00000000010504a7 in Slic3r::Arachne::SkeletalTrapezoidation::discretize(boost::polygon::voronoi_edge<double> const&, std::vector<Slic3r::Arachne::PolygonsSegmentIndex, std::allocator<Slic3r::Arachne::PolygonsSegmentIndex> > const&)
    ()
#3  0x000000000105285a in Slic3r::Arachne::SkeletalTrapezoidation::transferEdge(Slic3r::Point, Slic3r::Point, boost::polygon::voronoi_edge<double>&, Slic3r::Arachne::STHalfEdge*&, Slic3r::Point&, Slic3r::Point&, std::vector<Slic3r::Arachne::PolygonsSegmentIndex, std::allocator<Slic3r::Arachne::PolygonsSegmentIndex> > const&) ()
#4  0x000000000105dd11 in Slic3r::Arachne::SkeletalTrapezoidation::constructFromPolygons(std::vector<Slic3r::Polygon, tbb::detail::d1::scalable_allocator<Slic3r::Polygon> > const&) ()
#5  0x000000000105f2f8 in Slic3r::Arachne::SkeletalTrapezoidation::SkeletalTrapezoidation(std::vector<Slic3r::Polygon, tbb::detail::d1::scalable_allocator<Slic3r::Polygon> > const&, Slic3r::Arachne::BeadingStrategy const&, double, int, int, int, int) ()
#6  0x0000000000fb0866 in Slic3r::Arachne::WallToolPaths::generate() ()
#7  0x0000000000f5b142 in Slic3r::PerimeterGenerator::process_arachne(Slic3r::Pe--Type <RET> for more, q to quit, c to continue without paging--c
rimeterGenerator::Parameters const&, Slic3r::Surface const&, std::vector<Slic3r::ExPolygon, std::allocator<Slic3r::ExPolygon> > const*, std::vector<Slic3r::Polygon, tbb::detail::d1::scalable_allocator<Slic3r::Polygon> >&, Slic3r::ExtrusionEntityCollection&, Slic3r::ExtrusionEntityCollection&, std::vector<Slic3r::ExPolygon, std::allocator<Slic3r::ExPolygon> >&) ()
#8  0x0000000000f1bab1 in Slic3r::LayerRegion::make_perimeters(Slic3r::SurfaceCollection const&, std::vector<std::pair<Slic3r::IndexRange<unsigned int>, Slic3r::IndexRange<unsigned int> >, std::allocator<std::pair<Slic3r::IndexRange<unsigned int>, Slic3r::IndexRange<unsigned int> > > >&, std::vector<Slic3r::ExPolygon, std::allocator<Slic3r::ExPolygon> >&, std::vector<Slic3r::IndexRange<unsigned int>, std::allocator<Slic3r::IndexRange<unsigned int> > >&) ()
#9  0x0000000000f15e3a in Slic3r::Layer::make_perimeters() ()
#10 0x00000000008fdc1e in tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<unsigned long>, Slic3r::PrintObject::make_perimeters()::{lambda(tbb::detail::d1::blocked_range<unsigned long> const&)#2}, tbb::detail::d1::auto_partitioner const>::execute(tbb::detail::d1::execution_data&) ()
#11 0x00000000018d3963 in tbb::detail::r1::market::process(rml::job&) ()
#12 0x00000000018d0633 in tbb::detail::r1::rml::private_worker::thread_routine(void*) ()
#13 0x00007ffff6894ac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#14 0x00007ffff6926660 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) info reg
rax            0x1fa5              8101
rbx            0xffd5baab          4292197035
rcx            0x851f              34079
rdx            0x1                 1
rsi            0xfc0d4045          4228726853
rdi            0x30b091            3190929
rbp            0xffffffdb3291b998  0xffffffdb3291b998
rsp            0x7fff58bf84a0      0x7fff58bf84a0
r8             0xfff7fcee          4294442222
r9             0xffffffdecde356e6  -142574659866
r10            0x4094f6d151        277377110353
r11            0x534d42            5459266
r12            0xa0d7d             658813
r13            0x7fff58bf8698      140734682334872
r14            0xfffffffffffe5887  -108409
r15            0x7fff58bf8760      140734682335072
rip            0x7ffff6b8712c      0x7ffff6b8712c <__atan_fma+588>
eflags         0x10203             [ CF IF RF ]
cs             0x33                51
ss             0x2b                43
ds             0x0                 0
es             0x0                 0
fs             0x0                 0
gs             0x0                 0
(gdb)`

carrant, 2023-12-21:
```
Thread 68 "slic3r_tbb_10" received signal SIGFPE, Arithmetic exception.
[Switching to Thread 0x7fff58bfc640 (LWP 36155)]
0x00007ffff6b8712c in __atan_fma (x=0.087266460061073303) at ../sysdeps/ieee754/dbl-64/s_atan.c:97
97    ../sysdeps/ieee754/dbl-64/s_atan.c: No such file or directory.
(gdb) bt
#0  0x00007ffff6b8712c in __atan_fma (x=0.087266460061073303)
    at ../sysdeps/ieee754/dbl-64/s_atan.c:97
#1  0x000000000106992b in Slic3r::Arachne::VoronoiUtils::discretizeParabola(Slic3r::Point const&, Slic3r::Arachne::PolygonsSegmentIndex const&, Slic3r::Point, Slic3r::Point, int, float) ()
#2  0x00000000010504a7 in Slic3r::Arachne::SkeletalTrapezoidation::discretize(boost::polygon::voronoi_edge<double> const&, std::vector<Slic3r::Arachne::PolygonsSegmentIndex, std::allocator<Slic3r::Arachne::PolygonsSegmentIndex> > const&)
    ()
#3  0x000000000105285a in Slic3r::Arachne::SkeletalTrapezoidation::transferEdge(Slic3r::Point, Slic3r::Point, boost::polygon::voronoi_edge<double>&, Slic3r::Arachne::STHalfEdge*&, Slic3r::Point&, Slic3r::Point&, std::vector<Slic3r::Arachne::PolygonsSegmentIndex, std::allocator<Slic3r::Arachne::PolygonsSegmentIndex> > const&) ()
#4  0x000000000105dd11 in Slic3r::Arachne::SkeletalTrapezoidation::constructFromPolygons(std::vector<Slic3r::Polygon, tbb::detail::d1::scalable_allocator<Slic3r::Polygon> > const&) ()
#5  0x000000000105f2f8 in Slic3r::Arachne::SkeletalTrapezoidation::SkeletalTrapezoidation(std::vector<Slic3r::Polygon, tbb::detail::d1::scalable_allocator<Slic3r::Polygon> > const&, Slic3r::Arachne::BeadingStrategy const&, double, int, int, int, int) ()
#6  0x0000000000fb0866 in Slic3r::Arachne::WallToolPaths::generate() ()
#7  0x0000000000f5b142 in Slic3r::PerimeterGenerator::process_arachne(Slic3r::Pe--Type <RET> for more, q to quit, c to continue without paging--c
rimeterGenerator::Parameters const&, Slic3r::Surface const&, std::vector<Slic3r::ExPolygon, std::allocator<Slic3r::ExPolygon> > const*, std::vector<Slic3r::Polygon, tbb::detail::d1::scalable_allocator<Slic3r::Polygon> >&, Slic3r::ExtrusionEntityCollection&, Slic3r::ExtrusionEntityCollection&, std::vector<Slic3r::ExPolygon, std::allocator<Slic3r::ExPolygon> >&) ()
#8  0x0000000000f1bab1 in Slic3r::LayerRegion::make_perimeters(Slic3r::SurfaceCollection const&, std::vector<std::pair<Slic3r::IndexRange<unsigned int>, Slic3r::IndexRange<unsigned int> >, std::allocator<std::pair<Slic3r::IndexRange<unsigned int>, Slic3r::IndexRange<unsigned int> > > >&, std::vector<Slic3r::ExPolygon, std::allocator<Slic3r::ExPolygon> >&, std::vector<Slic3r::IndexRange<unsigned int>, std::allocator<Slic3r::IndexRange<unsigned int> > >&) ()
#9  0x0000000000f15e3a in Slic3r::Layer::make_perimeters() ()
#10 0x00000000008fdc1e in tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<unsigned long>, Slic3r::PrintObject::make_perimeters()::{lambda(tbb::detail::d1::blocked_range<unsigned long> const&)#2}, tbb::detail::d1::auto_partitioner const>::execute(tbb::detail::d1::execution_data&) ()
#11 0x00000000018d3963 in tbb::detail::r1::market::process(rml::job&) ()
#12 0x00000000018d0633 in tbb::detail::r1::rml::private_worker::thread_routine(void*) ()
#13 0x00007ffff6894ac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#14 0x00007ffff6926660 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) info reg
rax            0x1fa5              8101
rbx            0xffd5baab          4292197035
rcx            0x851f              34079
rdx            0x1                 1
rsi            0xfc0d4045          4228726853
rdi            0x30b091            3190929
rbp            0xffffffdb3291b998  0xffffffdb3291b998
rsp            0x7fff58bf84a0      0x7fff58bf84a0
r8             0xfff7fcee          4294442222
r9             0xffffffdecde356e6  -142574659866
r10            0x4094f6d151        277377110353
r11            0x534d42            5459266
r12            0xa0d7d             658813
r13            0x7fff58bf8698      140734682334872
r14            0xfffffffffffe5887  -108409
r15            0x7fff58bf8760      140734682335072
rip            0x7ffff6b8712c      0x7ffff6b8712c <__atan_fma+588>
eflags         0x10203             [ CF IF RF ]
cs             0x33                51
ss             0x2b                43
ds             0x0                 0
es             0x0                 0
fs             0x0                 0
gs             0x0                 0
(gdb)
```

lukasmatena, 2023-12-21:
Thanks. The backtrace shows that this is a duplicate of #11830. I am not exactly sure what the issue is. Could you please try [this build](https://slicerbuilds.prusa3d.com/PrusaSlicer-2.7.1+1-lm-atan-experiments-linux-x64-GTK3-g4a89241-202312211957.AppImage)? It is just a guess, but probably worth trying.

carrant, 2023-12-21:
That link doesn't resolve to a file.

[image]

carrant, 2023-12-24:
Little more info, I built prusa-slicer from git and still have the crash.
If I disable the E cores the crash does not occur.  I have not tried the reverse ( P cores offline)

System info
```
CPU - 
carrant@carranti9:~$ cpu-info
Packages:
	0: Intel 13th Gen Core i9-13900KS

carrant@carranti9:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy
```

Here are the cores before shutting them off -
```
carrant@carranti9:~/$ lscpu --all --extended
CPU NODE SOCKET CORE L1d:L1i:L2:L3 ONLINE    MAXMHZ   MINMHZ      MHZ
  0    0      0    0 0:0:0:0          yes 5600.0000 800.0000 5458.865
  1    0      0    0 0:0:0:0          yes 5600.0000 800.0000 3200.000
  2    0      0    1 4:4:1:0          yes 5600.0000 800.0000 5600.000
  3    0      0    1 4:4:1:0          yes 5600.0000 800.0000 3200.000
  4    0      0    2 8:8:2:0          yes 5600.0000 800.0000 5600.000
  5    0      0    2 8:8:2:0          yes 5600.0000 800.0000 3200.000
  6    0      0    3 12:12:3:0        yes 5600.0000 800.0000 5543.973
  7    0      0    3 12:12:3:0        yes 5600.0000 800.0000 3200.000
  8    0      0    4 16:16:4:0        yes 5900.0000 800.0000 5575.145
  9    0      0    4 16:16:4:0        yes 5900.0000 800.0000 3200.000
 10    0      0    5 20:20:5:0        yes 5800.0000 800.0000 5600.013
 11    0      0    5 20:20:5:0        yes 5800.0000 800.0000 3200.000
 12    0      0    6 24:24:6:0        yes 5800.0000 800.0000 5600.000
 13    0      0    6 24:24:6:0        yes 5800.0000 800.0000 5121.587
 14    0      0    7 28:28:7:0        yes 5800.0000 800.0000 5697.364
 15    0      0    7 28:28:7:0        yes 5800.0000 800.0000 3200.000
 16    0      0    8 32:32:8:0        yes 4600.0000 800.0000 4600.004
 17    0      0    9 33:33:8:0        yes 4600.0000 800.0000 4600.015
 18    0      0   10 34:34:8:0        yes 4600.0000 800.0000 3200.000
 19    0      0   11 35:35:8:0        yes 4600.0000 800.0000 4608.105
 20    0      0   12 36:36:9:0        yes 4600.0000 800.0000 4600.192
 21    0      0   13 37:37:9:0        yes 4600.0000 800.0000 4599.604
 22    0      0   14 38:38:9:0        yes 4600.0000 800.0000 4602.736
 23    0      0   15 39:39:9:0        yes 4600.0000 800.0000 4599.918
 24    0      0   16 40:40:10:0       yes 4600.0000 800.0000 4600.477
 25    0      0   17 41:41:10:0       yes 4600.0000 800.0000 3200.000
 26    0      0   18 42:42:10:0       yes 4600.0000 800.0000 4600.211
 27    0      0   19 43:43:10:0       yes 4600.0000 800.0000 3200.000
 28    0      0   20 44:44:11:0       yes 4600.0000 800.0000 4602.186
 29    0      0   21 45:45:11:0       yes 4600.0000 800.0000 3200.000
 30    0      0   22 46:46:11:0       yes 4600.0000 800.0000 4603.528
 31    0      0   23 47:47:11:0       yes 4600.0000 800.0000 3200.000
```

Here is what I used to shut off the E cores
```
echo 0 > /sys/devices/system/cpu/cpu8/online
echo 0 > /sys/devices/system/cpu/cpu9/online
echo 0 > /sys/devices/system/cpu/cpu10/online
echo 0 > /sys/devices/system/cpu/cpu11/online
echo 0 > /sys/devices/system/cpu/cpu12/online
echo 0 > /sys/devices/system/cpu/cpu13/online
echo 0 > /sys/devices/system/cpu/cpu14/online
echo 0 > /sys/devices/system/cpu/cpu15/online
echo 0 > /sys/devices/system/cpu/cpu16/online
echo 0 > /sys/devices/system/cpu/cpu17/online
echo 0 > /sys/devices/system/cpu/cpu18/online
echo 0 > /sys/devices/system/cpu/cpu19/online
[... 159 lines cut ...]
rbx            0x7fff3b6d72c0      140734190416576
rcx            0x7fffc82776b8      140736551417528
rdx            0x0                 0
rsi            0x7fff11b8d4b0      140733490713776
rdi            0x7ffee1d56578      140732687279480
rbp            0xfb5f6c01          0xfb5f6c01
rsp            0x7fff3b6d7200      0x7fff3b6d7200
r8             0x7fff3b6d72c0      140734190416576
r9             0x0                 0
r10            0xaaaaaaaaaaaaaaab  -6148914691236517205
r11            0x7ffff5ad6a24      140737315170852
r12            0x5555561a59f8      93825005148664
r13            0x7ffee1d5eed0      140732687314640
r14            0x7fff3b6d72c0      140734190416576
r15            0x7fff3b6d72a0      140734190416544
rip            0x7fff3b6d72c0      0x7fff3b6d72c0
eflags         0x10202             [ IF RF ]
cs             0x33                51
ss             0x2b                43
ds             0x0                 0
es             0x0                 0
fs             0x0                 0
gs             0x0                 0
(gdb)
```

lukasmatena, 2023-12-24:
Sorry, I edited my post and fixed the link.

carrant, 2023-12-24:
Thanks for the updated link!
Downloaded and ran it - same results.

closed 2026-09-29 by prusaslicer-bot[bot]