prusa3d/PrusaSlicer #11940: Dots on multicolor prints on Prusa XL

bobbyh41, 2023-12-18:
### Description of the bug

I am having dots end up on my multicolor prints on my XL(firmware 5.1.2) and I think it is caused by a little circular motion that happens between some layers, there is an example of this on layer 148 gcode lines 105141-105157.
[image]

https://github.com/prusa3d/PrusaSlicer/assets/93236040/6f93ea32-f0c2-44c3-9e94-f3762c1d76ba

[image]

### Project file & How to reproduce

[DotExample.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13709433/DotExample.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Mac 14.1.2

### Printer model

Prusa XL 5 head 5.1.2

Jan-Soustruznik, 2023-12-19:
Hello, @bobbyh41 ,

thank you for your report. We will try it to fix in some next release. Have a nice day 

add to ticked: SPE-2085

bobbyh41, 2023-12-19:
I just realized that the original time I printed the model that had the dots I had the seam position in the rear, and that might change things, that might not be what is causing the dots, but the dots are there on many prints.

bobbyh41, 2023-12-27:
Here is another example of it happening again I think, it might be hard to see the little red circles in the pic in the white part, but they are there.
[image]
[image]

lukasmatena, 2024-02-09:
Fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2). Closing.

closed 2024-02-09 by lukasmatena