prusa3d/PrusaSlicer #11946: Bug generating G-code 90%

vo258jta, 2023-12-20:
### Description of the bug

With big STL file (complicated), its often to stop at 90% generating G-code, and nothing happens in hour...
[image]

### Project file & How to reproduce

this file is too big

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Verze 2.7.1+win64

### Operating system

Win 11

### Printer model

Ender 3 Pro Klipper

Hobbit0, 2024-01-03:
I have the same issue, 
2.6.1 handles big files easily, 2.7.1 stops at 90% and crashes if i try to stop slicing or change any setting. 
Also seems to me like 2.6.1 uses more available resources (which is good)
[image]
[image]

rossiam, 2024-09-02:
This is also happening for me on an M1 MacBook with 64GB RAM, PrusaSlicer Version 2.8.0+MacOS-arm64, though I am able to actually get the slicing done if I'm patient enough. I'm using the default `0.20mm QUALITY` profile with a 0.4mm nozzle.

I'm slicing this model:

https://www.printables.com/model/37388-twisted-scoping-toy

* The tiny version hangs for a short time at 90%.
* The small version hangs for a longer.
* The normal version takes about 10 minutes.
* The taller version took about an hour. (Cura slices this version in < 2 seconds. 😄)

PrusaSlicer shows as responding in the "Force Quit Applications" dialog while it's slicing unless I try to cancel it or do anything else. (If I click on anything while it's stuck at 90% it changes to "Not Responding" and I get a busy cursor. I've never been patient enough to see if the UI would become responsive at this point.)

PrusaSlicer uses about 50-75% of a CPU while it is stuck at 90%.

I have sliced this model with PrusaSlicer in the past with no problems.

closed 2026-09-29 by prusaslicer-bot[bot]