prusa3d/PrusaSlicer #11948: There is a way to completely eliminate the visible seam.

vgdh, 2023-12-20:
# Hide the Seam Completely
The whitepaper describes how to completely eliminate the visible seam.

## Explanation
Every closed loop will eventually have a visible seam. However, if there is no specific point where the line ends, there is no seam, right?
1. Begin the line with a height of 0 and gradually increase the height of the nozzle and the line over some length until it reaches the full height of the line.
2. Move through the loop until reaching the end of the line, but continue extruding the filament due to the initial zero line height. Systematically reduce the line's height while maintaining the same nozzle height until reaching the end of the starting slope.
3. Move on to the next perimeter or infill.

That's it. The algorithm is remarkably simple, though the result is astonishingly good. 
It is not an easy task to post-process already existing GCODE files, but if it works at the slicer level, it will work very well.

## The python code
The repository includes an [example](https://github.com/vgdh/seam-hiding-whitepaper/blob/main/main.py) demonstrating a basic implementation of the algorithm for printing a ring. This implementation allows specifying parameters such as the number of lines in the ring, layer height, line width, speed, etc.

Also there is an [example](https://github.com/vgdh/seam-hiding-whitepaper/blob/main/nozzle%200.4%20line%200.3x0.8.gcode) of gcode file for 0.4 nozzle line 0.3x0.6 layer height 0.3

## Photos of the Seam Details on the Printed Example.
The first layer (the top on the first photo) was printed without the algorithm to get better adhesion.
[image]
[image]
[image]

In the photo below you can see there is no gap between the lines.
[image]
[image]
[image]
[image]
[image]

neophyl, 2023-12-20:
Isn't this basically the same as this https://github.com/prusa3d/PrusaSlicer/issues/11621  ?  Which is already open as a feature request.

MichaelJLew, 2023-12-20:
This is an  excellent idea and I support it strongly. I have also posted it, as neophyl noted. I don't suppose that having two posts on the same topic is a problem, but you should probably comment on the earlier one, and perhaps link your Python code there as well.

vgdh, 2023-12-21:
unfortunately I don't find [MichaelJLew](https://github.com/MichaelJLew) post when I had searched for this method. And sadly I guess the prusa team don't really interested in this method of filament line joint.

MichaelJLew, 2023-12-21:
> the prusa team don't really interested

Don't seem interested _yet_. This method is a major advance that can be used any time it is sensible to print the outer perimeter first. I am confident that it will find some interest from the Prusa team when they find time to look at it. We need to keep promoting it.

vgdh, 2023-12-26:
I have written some post-processing [script](https://github.com/vgdh/seam-slope-postprocessor/blob/main/postprocessor_seam_slope.py). Feel free to test, modify, or contribute.

vgdh, 2024-01-01:
Now the script processes only the outer perimeters, printing the inner ones as usual.

vgdh, 2024-01-01:
bender likes it very much 😁
[image]
[image]

sirgrej, 2024-01-06:
Is it only working with specific slicer settings? I mean: is it only working with 0.3 layer height? Yesterday I tried your script with the default mk4 settings of prusaalicer 2.7.1 and input shaping @ 0.2 layer height. Can't see any improvements of the seam

vgdh, 2024-01-06:
> Is it only working with specific slicer settings? I mean: is it only working with 0.3 layer height? Yesterday I tried your script with the default mk4 settings of prusaalicer 2.7.1 and input shaping @ 0.2 layer height. Can't see any improvements of the seam

0.3 layer height is recommended, acceleration on external perimeter should be lower than internal.

vgdh, 2024-01-06:
[sirgrej](https://github.com/sirgrej), there is the full description [readme](https://github.com/vgdh/seam-slope-postprocessor)

vgdh, 2024-01-28:
there is some progress on this feature.
[PR](https://github.com/SoftFever/OrcaSlicer/actions/runs/7683516898?pr=3839)