prusa3d/PrusaSlicer #11952: Save view settings in gcode view (like travel moves, seam etc)

KlausHans, 2023-12-20:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when i have to set my preferred view settings in the gcode view again and again.
[image]

**Describe the solution you'd like**
The chosen view settings are the same like i set them before when i reopen Prusa Slicer (in other words the settings are saved).

**Describe alternatives you've considered**
Setting them by hand again and again is possible but tedious.

**Additional context**
I think this feature request may exists already because this is a quite basic thing, in my opinion. But i did not found a similar issue.

Software used:
Ubuntu 23.10
PrusaSlicer 2.7.1, Appimage

neophyl, 2023-12-20:
Please check for existing requests before opening new ones. This has been requested multiple times before. For example https://github.com/prusa3d/PrusaSlicer/issues/9468 or how about https://github.com/prusa3d/PrusaSlicer/issues/8418

KlausHans, 2023-12-20:
I did check but didn't found those. Thank you. 
I close this.

closed 2023-12-20 by KlausHans