prusa3d/PrusaSlicer #11956: Saving of G-Code export path in project file

Jeggo65, 2023-12-21:
**Is your feature request related to a problem? Please describe.**
It is not a problem but inconvenient to enter the export path with ever project change

**Describe the solution you'd like**
The project file (.3MF) does not store the export path of the G-Code. If you use a structured storage on the printer, you will have to change the path by typing the new destination path. 

**Describe how it would work**
Export path for G-Code is stored in the project file and may be different for each project.

**Describe alternatives you've considered**
Edit the path by hand each time.

**Additional context**
N.A.

neophyl, 2023-12-21:
How do you have these preferences set ?
[image]

For me it exports gcode files to the same directory where my project is saved.  So File>Save Project As to save my 3mf with all the settings and models etc and then when generating gcode it always saves to the same location as the 3mf.  If I load the project back in later then it knows where the project came from and exports gcode there.

Jeggo65, 2023-12-21:
Maybe it was not clear. I export the G-Code either to OctoPrint or now to PrusaLink. So directly to the OctoPrint server or to the USB drive of the printer. In both cases the path is not saved in the project and has to be edited every time.
The option "remember output directory" is set. But this seems to apply only for pc network (not printer). Or is it possible to connect the USB drive of the printer via WebDAV or something like this as a network share to a local computer?

closed 2026-09-29 by prusaslicer-bot[bot]