prusa3d/PrusaSlicer #11958: Random glitches when adding SVG shape

ronperek, 2023-12-21:
### Description of the bug

When I add an SVG file to be printed, random parts of SVG are replaced with a rectangle.

If I modify a part of the SVG file, without modifying the affected area, the parts that are replaced with a rectangle change. That's why the problem feels random rather than associated with a particular part of the SVG.

### Project file & How to reproduce

Example SVG file:
[image]

Project:
[pinout.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13745992/pinout.3mf.zip)

Screenshot:
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.1+flathub.org

### Operating system

Linux Mint 21.2 Cinnamon

### Printer model

MK3S

kubispe1, 2023-12-22:
I forwarded your issue to Develeper to check it, but I think SVG is corrupted in this way:
- The final shape contains a self-intersection or multiple points with the same coordinates
So am closing the issue. In case we will reopen it. Thanks

closed 2023-12-22 by kubispe1

ronperek, 2023-12-22:
So why does it affect only one letter "S" when all the "S" letters are the same?
And why if I do something like canvas size change, the affected letter starts working and some other letters break?

Jony01, 2023-12-22:
In SVG is multiple times issue with shape. 
PrusaSlicer is able automatically heal a lot of them, but still there are some cases, we can't prevent.
Long story short: By [ IEEE 754](https://en.wikipedia.org/wiki/IEEE_754) floating point number precision is more accurate around zero.

I see that letters are created by segments:
e.g. character 'S' I attached as an image is consist of 11 objects
[image]

Jony01, 2023-12-22:
For heal this particalar case you could use SVG editor (I suggest to use [Inkscape](https://inkscape.org/)) convert your stroke to path(in inkscape: Path/StrokeToPath --> ctrl+alt+c ) and than unify path(Path/Union --> ctrl++).
[image]
[image]

ronperek, 2023-12-22:
My file is displayed correctly in Inkscape, Firefox, even in 2D preview in Prusa Slicer. Is it really corrupted?

If the Prusa Slicer can deal with most of the letters and the problems are caused by floating point rounding, maybe it is possible to tweak the slicer to deal with all of them?

It would be nice to be able to just drop an SVG file that correctly displays in 2D to Prusa Slicer without any preprocessing.

The commands in Inkscape you suggested did something weird to the shape of the letters.
[image]

Jony01, 2023-12-22:
Yes, that is the isssue which PS must deal with too. Inkscape solve it by the result you see. I also disagree that this result is what one would like to achieve.

2d editor do not need closed linearizer loops of contour but mesh consist of triangles do.

ronperek, 2023-12-22:
So what is the final decision?
Do you plan to improve the slicer to be able to deal with the file I provided?

Jony01, 2023-12-22:
My final decision is that I don't know better way to deal with this. If you can imagine better solution, volunteering is welcome. 
I do not plan to change current implementation.
My suggestion is: "For 3d embossing use SVG without overlaping to prevent issues like this." 
It is rarely appear, only overlaped almost same curves can cause it.

Jony01, 2023-12-22:
Another way to fix SVG is unify points of line segments.
[image]
[image]
[image]

ronperek, 2023-12-22:
I tried the function you mentioned with the sample file I uploaded here earlier and it worked.
Unfortunately, it doesn't help with the full SVG file I need to print. It just changes which letters are printed correctly and which fail. Say hello to the randomness I mentioned :(

Do you have any suggestions which Inkscape functions to use to make it work in the slicer and avoid text deformation in the process?

Here is the full file I need to print:
[image]

Jony01, 2023-12-23:
Another way to union shape is export raster image and than trace bitmap.

Jony01, 2023-12-23:
I disagree that it is random. For the SVG file behavior is still the same.

ronperek, 2023-12-23:
I'm not an expert in SVG format or slicing. I mean that for me, it seems like a random failure.

Look at the three different SVG files. I just changed the canvas size and moved the text around without any advanced transformation of the SVG objects. Each SVG file causes different letters to fail. That is what seems random to me.

That also led me to thinking that this is a bug and not a problem with a particular part of the SVG file. It's still not 100% clear to me why the Prusa Slicer can deal correctly with the SVG letter placed in one part of the canvas end fails when it's moved to other position.

1:
[image]
[image]

2:
[image]
[image]

3:
[image]
[image]

ronperek, 2023-12-25:
Merry Christmas Jony01. Thank you for your help. I have some extra notes for those who fight with the same issue. My current workflow to get it working is:

1. Load the initial file to Prusa Slicer to see which SVG objects fail and are changed to rectangles.
2. Manually fix only the affected SVG objects using the "unify points of line segments" Inkscape technique you mentioned.
3. Reload the SVG file in PS. Bonus: Now the file processing is much faster (lets say something like 10x faster).
4. Don't touch the SVG again to avoid other objects failing.

Unfortunately, the "unify points of line segments" method needs selecting nodes letter by letter. That's why I only fix the affected objects. When all the nodes are selected, Inkscape tries to make a loop from every letter xD.

If you find some spare time, here's an idea for a fix:
Prusa Slicer already fixes some STL issues on import. Maybe it can fix this issue with SVG?
The function can scan for points that are less than some small distance from each other. Maybe less than 0.001mm?
And then use the same algorithm that Inkscape uses to unify them.
Then the PS can proceed with the further processing of the file.