prusa3d/PrusaSlicer #11961: Auto-Arrange group by height

cadacus, 2023-12-21:
**Is your feature request related to a problem? Please describe.**
Not really a problem, but I think it would speed up printing.

**Describe the solution you'd like**
When the arrange button is pressed, the height (Z-axis) of each item would be taken into account and similar height items grouped near each other. The theory is there would be less movement between pieces as shorted items are completed.

**Describe how it would work**
I would assume each item on the build plate would be sorted by height and then laid out in that order. I don't know how the process is done now, but that would seem to be a way to keep similar height items near each other.

**Describe alternatives you've considered**
I didn't think of this until I watched the printer head have to move across the build plate between items because the shorter items had been finished. If I had realized it, I probably would have moved the higher items closer together manually.

**Additional context**
Add any other context or paint drawing about the feature request here.

kubispe1, 2023-12-22:
Thanks for idea.

Tupson444, 2026-09-18:
still relevant

Jan-Soustruznik, 2026-10-21:
A similar feature request already exists. This request has been merge to keep topics organized.

closed 2026-10-21 by Jan-Soustruznik