prusa3d/PrusaSlicer #11964: [FR] Implement Cura "Make Overhangs Printable" feature.

vovodroid, 2023-12-22:
See https://forum.prusa3d.com/forum/prusaslicer/make-overhang-printable-option/  discussion.

[image]

[image]

eoyilmaz, 2023-12-22:
I think OrcaSlicer already implemented this feature: https://github.com/SoftFever/OrcaSlicer/discussions/1319

maybe it is easy to back port...

vovodroid, 2023-12-22:
Yes, this one: https://github.com/SoftFever/OrcaSlicer/pull/1615

I'll try to merge this code and see whether it at least compiles.

kubispe1, 2023-12-22:
Nice, we will check it. Thanks for the info.

vovodroid, 2023-12-22:
I copied code from Orca and build, here is result:

[image]

[image]

@kubispe1 would you like me to open PR?

eGit, 2024-11-11:
When will this finally be available? Is there any timetable?

closed 2026-09-29 by prusaslicer-bot[bot]