prusa3d/PrusaSlicer #11965: [FR] Implement One Top Perimeter Feature from Superslicer (existing PR)

ndanyluk, 2023-12-22:
**Is your feature request related to a problem? Please describe.**
Not so much a problem, as an improvement.  Currently the default behavior on PS is to always have the same number of perimeters on all layers.  This is structurally fine.  However, in SS, there exists a feature to only have one perimeter on top/bottom.  This leads to beautiful looking parts that retain nearly all of their structural strength and have a uniform surface on top/bottom.

**Describe the solution you'd like**
I would like the following PR to be considered for inclusion in a future PS release: #10648 

**Describe how it would work**
The user would have a checkbox in their slicer profile that allows for one top/bottom perimeter.  The implementation details are in the PR, but the net effect for the user is that clicking this box would modify the sliced gcode such that top surfaces & bottom surfaces only have a single perimeter.  

**Describe alternatives you've considered**
Today in PS you can get close to this behavior with height range modifiers to restrict the number of perimeters.  However, the limitation with this is that it can only be done on the absolute top/bottom layers.  If there is an intermediate surface (for example a recessed top surface in a part), this workaround is impossible to implement for that surface without compromising the structural integrity of the part overall (since height range modifiers would remove the perimeters on the full layer vs just the exposed surface).

**Additional context**
[image]
[image]
Example of sliced filament card on PS 2.7

[image]
[image]
Example of sliced filament card on SS 2.4

ndanyluk, 2023-12-22:
To be clear, I love the work the PS team does.  This is one of the only things that keeps me from switching back from Superslicer and I wish that the team would give the feature a look and comment definitively on whether it is something that we will ever see in PS.

kubispe1, 2023-12-29:
I appreciate your comment and description. We have it on our list, but it is not yet in progress. We apologize for the delay.
I am closing this issue. Please feel free to comment on the original issues ( #7986, #2151), or better yet, in the mentioned PR ( #10648).
Thanks and have a nice day

closed 2023-12-29 by kubispe1