prusa3d/PrusaSlicer #11966: Slicer adds .2mm layer

taz43mca, 2023-12-22:
### Description of the bug

I have designed a cylinder (tower) and when bringing the 3MF or STL file into slicer it looks OK. Then when I slice it I get one or more .2 mm layer added to the inside of the cylinder. I have made sure that the files are OK in my windows 3D builder software and in the slicer. No errors shown in either. I have tried all kinds of modifications to the file and always get the same problem.  I have never had this problem before using 3d Builder, STL and 3MF files. I have not tried installing an older version of Prusa slicer and running the file. Attached is a zip file of the 3MF file and screen shots of the pre and post slicer bed shots.
[image]
[image]

### Project file & How to reproduce

[LG Lighthouse Tower F 3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13756505/LG.Lighthouse.Tower.F.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

windows 11 Pro

### Printer model

Prusa Mk 2.5S

neophyl, 2023-12-23:
There is something funky about that 3mf.  It appears to have not been saved from Prusa Slicer as none of the profile info is there, it just uses whatever profiles you had loaded so it just looks like geometry/model data and not a Prusa Slicer project.

Secondly when I open it I get the choice to load it either as a single object made up of 2 parts or 2 objects. One of the objects is a very thin part that seems to go through the other vertically.
[image]

As such it exceeds the size of the mk3 print bed.  If I remove that part so that the main object falls to the bed and becomes printable and slice with some default profiles then it slices normally (some warnings about floating bridge anchors).
[image]

This is with 2.7.1 on Windows 10.  

If I export the model to Blender it confirms it is manifold but it has ALOT of intersecting faces, 560339 of them to be precise.  Blender also shows that extraneous geometry sticking above and below.

taz43mca, 2023-12-23:
I have none of the problems you had. I sliced it in Prusa Slicer 2.7.1 as stated. I just sliced it in Prusa Slicer 2.5.2 and it did not have the problem of the .2mm layer. As shown in the pictures, it does fit on the MK3 bed. It was sliced for the Mk2.5S.
Your results are strange as I have had none of those problems. I had printed the corrupt files that were attached and I get the .2mm layer which I cut out and clean up the area where it was attached. 
Since I had no problem with the file sliced with 2.5.2 slicer, I may just do my slicing with that version and print from these files.
Does not solve the problem but does allow me to continue with my printing. Obviously cannot do texting that is available in 2.7.1 so will have to do in 3D blender and then save for slicing in Prusa slicer version 2.5.2.

taz43mca, 2023-12-23:
[20231212_102916.zip](https://github.com/prusa3d/PrusaSlicer/files/13759906/20231212_102916.zip)
Copy of completed print assembled.

taz43mca, 2023-12-27:
Very strange as it was designed in Windows 11 3D Builder, saved as a 3MF file, then saved as an STL file (same 3d builder software), STL file repaired in 3D builder; sliced in Prusa Slicer 2.7.1 (also tried 2.5.0 with same results); note that it does not put in horizontal .2mm plate in same place always and sometimes puts in more than one. Have sliced in Prusa 2.7.1 and 2.5.1 of both the 3mf and stl files. It says there are over a million triangles. Could it be a problem with the slicer when used in windows 11 Pro?
 You may have noticed that I also posted another unrelated problem, or maybe related. Using latest firmware for my Mk 2.5S I can no longer unload filament when I do a pause. I have done this for years with no problem but now when I pause the printer and look at the directories, there is no unload filament option.

 

image.png (view on web)

image.png (view on web)

neophyl, 2023-12-27:
I don't know the cause, all I know is when the model is examined in Blender there are many issues, for example this is what one of your windows looks like when zoomed in-
[image]

Now virtually none of that geometry will print/slice as its all thin faces so wont effect your output.  However its knocking up the face count considerably and depending on other things like part orientation could cause issues such as the phantom layers.  Also functions like using one of the paint tools on that area would give it fits.  And there geo like that All Over the model.  
Another example just from the exterior wall 
[image]
The interior is as bad if not worse than the exterior.

If you look at the pictures you posted in your very first post you can see 'streaky' lines vertically on both the green model and the slice.  That's all down to these extra faces so the issue is there in your original model.  Whatever process you used to assemble it looks like its caused all sorts of modelling issues.  Its not what I would describe as 'clean' model geometry.

kubispe1, 2023-12-29:
Hi, I can see all of the mentioned model abnormalities. I would guess that your issues with model precision are related to these things. If you could try a different export, @taz43mca ?
I am closing this issue for now. If you find that it is a problem with the PS, we will reopen it.

closed 2023-12-29 by kubispe1

taz43mca, 2024-01-03:
I know it is closed but I redid the item and now have no problem. Must have been how I designed it. It also prints faster.