prusa3d/PrusaSlicer #11971: Prusa Slicer mit WLAN für AnkerMake M5C

Detlef1965-Miz, 2023-12-24:
### Description of the bug

Hallo,

ich komme aus Germany können Sie nicht mal in den Prusa Slicer eine Funktion miteinbauen wo man der Datei STL, die man mit dem Slicer bearbeitet hat dann direkt per WLAN auf den Drucker senden kann ?
Ich habe leider kein Prusa 3D Drucker, sondern einen AnkerMake M5C.

### Project file & How to reproduce

Hallo,

ich komme aus Germany können Sie nicht mal in den Prusa Slicer eine Funktion miteinbauen wo man der Datei STL, die man mit dem Slicer bearbeitet hat dann direkt per WLAN auf den Drucker senden kann ?
Ich habe leider kein Prusa 3D Drucker, sondern einen AnkerMake M5C.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusa Slicer 2.7.

### Operating system

Windows 10

### Printer model

AnkerMake M5C

sarusani, 2023-12-24:
Prusa Slicer unterstützt das hochladen von gcode auf einen Drucker schon seit Jahren. Das Problem ist, dass Anker dies nicht unterstützt.

[Hier](https://github.com/Ankermgmt/ankermake-m5-protocol) gibt's ein Projekt das dies ermöglichen sollte.

Detlef1965-Miz, 2023-12-25:
hättest Du für mich ein fertiges Programm dazu ?

closed 2026-09-29 by prusaslicer-bot[bot]