prusa3d/PrusaSlicer #11976: Raft is not under Model

inspectionsbybob, 2023-12-24:
### Description of the bug

I was having problems with adhesion so extended my raft so it would interconnect all of my supports. The Raft extended quite a ways out from my Model IN THE WRONG DIRECTION.   I had to make the raft even larger to get it to print.
[Goblin_librarian.zip](https://github.com/prusa3d/PrusaSlicer/files/13762857/Goblin_librarian.zip)

### Project file & How to reproduce

Look 3MF enclosed and Images

There is a very large raft, with nothing over it to the left of my Goblin.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Win11

### Printer model

Various

sarusani, 2023-12-25:
The raft didn't expand in the wrong direction. A raft just goes outwards from the point where your model touches the print bed, it doesn't care where your supports are. Interconnecting your supports is not it's intended use.

To help with your support adhesion try the "Support material"->"First layer density" & "First layer expansion" settings.
Try to set the density to 100% and expansion to 10mm.

inspectionsbybob, 2023-12-25:
Thanks...

More importantly, happy holliday!  Its my hobby, somI am estatic that
someone was watching the support box on a holiday.

Thanks!

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

closed 2026-09-29 by prusaslicer-bot[bot]