prusa3d/PrusaSlicer #11980: support blocker not working after mirroring a part

otaolafranc, 2023-12-25:
### Description of the bug

If we mirror a part and then use paint support blocker to use support in desired section, during the support creation this support blockers are ignored

### Project file & How to reproduce

[project.zip](https://github.com/prusa3d/PrusaSlicer/files/13766658/project.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.1+linux-x64-GTK3-202312121425

### Operating system

linux Tuxedo OS 2

### Printer model

mk4

miklosbecsei, 2024-03-06:
I ran into the same issue today, would be amazing to have a fix for this.

Nextnx, 2025-02-18:
Found this today, so still an issue in version 2.9.0
If you use a block support box on the the left handed object it's also gets ignored.
I'm using "Support on build plate" with organic support.

otaolafranc, 2026-09-02:
> This issues has been labeled **legacy**. It means that it has been created before the release of _PrusaSlicer 3.0.0 alpha_ version. In an attempt to clean up the GitHub issues tracker any **legacy** [...]

this issue is still happening