prusa3d/PrusaSlicer #11981: M106 M107 M106 In Quick Succession Causes Part Cooling Fan To Repeatedly Go Into "Kick Start" Mode

Gunner087, 2023-12-25:
### Description of the bug

When using "dynamic fan speeds" M106 and M107 commands are created very close together, resulting in the printer firmware "kick starting" the fan constantly every time an M106 command is called after an M107 command. This results in the fan pulsing at a high speed which is usually above what was requested by the slicer.
[image]

The video shows how I can replicate the issue by simply spamming M106 and M107 at the printer using the console in Klipper and then shows how only sending M106 commands fixes the issue. I have my "kick start time" set to 0.5 seconds. Not sure if you would consider this a slicer bug or if this is something that should be fixed on the printer firmware side so I will also open this issue on the Klipper GitHub also.
[image]

https://github.com/prusa3d/PrusaSlicer/assets/65746045/7925f8f2-c399-45ab-a62d-3dad2d2fbb8a

### Project file & How to reproduce

[M106 M107 Issue.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13766818/M106.M107.Issue.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

macOS Sonoma 14.2.1

### Printer model

Voron 2.4 running Klipper v0.11.0-304

gman10192, 2024-01-10:
Spamming of M107 was found to be the cause of my printer locking X and Y while Z would still move as if simulating the print happening.

Jubilee build running a Duet 3 Main Board 6HC and Duet 3 Tool Boards 1LC for each of four tools.  Running Duet/RRF firmware 3.5.0 rc2.

cdheiser, 2024-01-21:
I spent half the day trying to chase down why my part cooling fan starts oscillating uncontrollably even after a print is finished.  I can confirm that the issue described here has the exact same symptoms that I've been chasing where disabling dynamic fan speeds resolves the issue.

I'm running Klipper v0.12.0-60

bobjackman, 2024-01-23:
Guys, please upvote this if you're experiencing it.

bobjackman, 2024-02-01:
related: https://github.com/prusa3d/PrusaSlicer/issues/11856

4res0, 2024-03-09:
Same problem.
PrusaSlicer 2.7.2
Klipper  v0.12.0-102

DementevAlex, 2024-03-17:
Also having this issue on Ender 3 with Klipper. 
Just printed a model with multiple bridges in a row. The fan stopped spamming almost at the end of a whole layer.
[image]
I think there should be some kind of averaging algorithm preventing fan to change speed faster than its `kick_start_duration` (which should be setup in slicer to be equal the value from klipper).

awabom, 2024-07-04:
Still happening in PS 2.8.0

Felix14-v2, 2024-08-07:
In my scenario, this issue causes my fan to complete deadlock on overhangs – it just can't start cooling after that.