prusa3d/PrusaSlicer #11985: Access Violation with organic supports

natn, 2023-12-27:
### Description of the bug

With the attached model, Prusa Slicer 2.7.1 crashes silently when organic supports are enabled and I attempt to slice.

The following appears in the event log:

Log Name:      Application
Source:        Application Error
Date:          12/27/2023 4:45:11 PM
Event ID:      1000
Task Category: Application Crashing Events
Level:         Error
Keywords:      
User:          XXXXX
Computer:      XXXXX
Description:
Faulting application name: prusa-slicer.exe, version: 2.7.1.0, time stamp: 0x657872dd
Faulting module name: PrusaSlicer.dll, version: 0.0.0.0, time stamp: 0x657872c8
Exception code: 0xc0000005
Fault offset: 0x000000000073ad2a
Faulting process id: 0x0x119C
Faulting application start time: 0x0x1DA39163A47A5E3
Faulting application path: C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe
Faulting module path: C:\Program Files\Prusa3D\PrusaSlicer\PrusaSlicer.dll
Report Id: 75d0931b-097a-4a40-95a5-5ce764c8103e
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" Guid="{a0e9b465-b939-57d7-b27d-95d8e925ff57}" />
    <EventID>1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x8000000000000000</Keywords>
    <TimeCreated SystemTime="2023-12-27T22:45:11.1962654Z" />
    <EventRecordID>76334</EventRecordID>
    <Correlation />
    <Execution ProcessID="15808" ThreadID="35588" />
    <Channel>Application</Channel>
    <Computer>NATN11</Computer>
    <Security UserID="S-1-5-21-874252260-4016202329-143669503-1001" />
  </System>
  <EventData>
    <Data Name="AppName">prusa-slicer.exe</Data>
    <Data Name="AppVersion">2.7.1.0</Data>
    <Data Name="AppTimeStamp">657872dd</Data>
    <Data Name="ModuleName">PrusaSlicer.dll</Data>
    <Data Name="ModuleVersion">0.0.0.0</Data>
    <Data Name="ModuleTimeStamp">657872c8</Data>
    <Data Name="ExceptionCode">c0000005</Data>
    <Data Name="FaultingOffset">000000000073ad2a</Data>
    <Data Name="ProcessId">0x119c</Data>
    <Data Name="ProcessCreationTime">0x1da39163a47a5e3</Data>
    <Data Name="AppPath">C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe</Data>
    <Data Name="ModulePath">C:\Program Files\Prusa3D\PrusaSlicer\PrusaSlicer.dll</Data>
    <Data Name="IntegratorReportId">75d0931b-097a-4a40-95a5-5ce764c8103e</Data>
    <Data Name="PackageFullName">
    </Data>
    <Data Name="PackageRelativeAppId">
    </Data>
  </EventData>
</Event>

### Project file & How to reproduce

[Wall Mount base - pre error.zip](https://github.com/prusa3d/PrusaSlicer/files/13781975/Wall.Mount.base.-.pre.error.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.1+win64

### Operating system

Edition	Windows 11 Pro Version	23H2 Installed on	‎9/‎27/‎2022 OS build	22631.2861 Experience	Windows Feature Experience Pack 1000.22681.1000.0

### Printer model

Original Prusa i3 MK4S+, w/ Rivo 6 hotend

neophyl, 2023-12-27:
Might be a win11 thing as it opens and slices fine on my Win10 machine with the same PS version.

natn, 2023-12-28:
I reloaded the 3mf file and it sliced fine.  I then reopened PS and loaded the STL, changed the support settings to organic, and it crashed when I tried to slice.  I then reloaded the 3mf and it then crashed when I attempted to slice.   I  don't yet know how to reproduce the success case consistently, but I have successfully sliced about on about 25% of the attempts.

natn, 2023-12-28:
Another note.  This does not occur when using grid supports and also affects some other models I have printed on previous versions of PS.

### I reverted to PS 2.6.1, and all models I have tested now slice consistently and correctly with organic supports.

Skirfir, 2023-12-30:
I have the same issue. It appears to me that this is indeed somehow related to Windows since I also have crashes in other programs as well. Though Prusa slicer produces the error more consistently than other programs. A couple of things I noticed is that it is not related to the version for me. When I use 2.7.0 or 2.6.0 it also crashes. It seems to be related to the size of the G-code simple objects with organic supports work fine for me but more complex ones do not. It also works when I run Windows in safe mode.

neophyl, 2023-12-30:
If it happens with other programs then the first thing I would check just to rule it out would be your pc's memory.  Run one of the memory checker programs that can put it through its paces and stress test it.

Skirfir, 2023-12-30:
I did that it didn't find anything.

kubispe1, 2024-01-02:
Hi, for a first time I cannot reproduce this issue for WIN10. I will forward it.
Thanks

lukasmatena, 2024-01-02:
@Skirfir @natn What processor models do you guys have? Could this be related to #11845?

Skirfir, 2024-01-02:
@lukasmatena I did try the fix described there since I also have an I9 13900K but it didn't help. But I tried something else just now and it turns out that one of my CPU cores is malfunctioning. So it isn't really related to Prusa slicer but thanks anyway.

natn, 2024-01-02:
> @Skirfir @natn What processor models do you guys have? Could this be related to #11845?

i9-13900K 128G RAM

lukasmatena, 2024-01-03:
Thanks. I am closing this in favor of #11845. I suspect it is indeed related.

closed 2024-01-03 by lukasmatena