prusa3d/PrusaSlicer #11987: [Feature request] Allow multimaterial importing for SVG files

pgilfernandez, 2023-12-28:
**Is your feature request related to a problem? Please describe.**
When printing multi-part (material or color) you usually load several STL files to match the materials (colors) you want to specify in the slicer. When you load them at once, there is a popup window that asks you if it's a multi-part object, if you set "yes" the volumes are imported taking in count their origins so that the objects are placed correctly among them.

**Describe the solution you'd like**
I'm asking for a similar feature but for SVG importing so that you can import several SVG files that you want to extrude and print as a multimaterial (color) object.

**Describe how it would work**
When loading SVG files, you could select not just one file but as many as you want. If several and you have a MMU printer then it asks you if it is a multimaterial print.

**Describe alternatives you've considered**
Allow to do it with the regular "import" feature, now it allows you to import several SVGs at a time, it even asks you if it's a multi-part object but their original XY origins aren't taken in count and they are just imported using their "bounding box center".

**Additional context**
(an example)
1) this is the vector design (for instance in Inkscape), the "assembled" design is the one that I would like to import. The "parts" just show the real part shapes.
[image]

2) this is how they get imported now, that is, the **no desired behavior** (I used the normal "import" feature, it asked me if they are a multi-part object) but as you see it didn't work as expected:
[image]

3) this is what I **expected to happen**:
[image]

4) so that later I can even play with heights to tweak the final print:
[image]

I hope you find the feature useful as it opens new opportunities for people with limited or no 3D designing skills but if they know how to do vector graphics then they can take advantage of this feature.
=D

Thanks!!

Jony01, 2024-01-02:
I agree that, SVG import should be extended for options multi parts/color/height. But I personally disagree that selecting multi files is the solution.
From my point of view: Design should be stored as one SVG file for future edit. And Settings (color, height, ...) for individual parts should be inside SVG gizmo.

pgilfernandez, 2024-01-04:
Hi @Jony01,
I like your suggestions, they let users/designers simplify the workflow but, in the way, make it more difficult to developers to add the feature. BTW I have the feeling this is the way to go, so I fully agree with you.

So, to make it clear: users/designers would just need to work on one SVG file with  more than one shape, when importing it into PrusaSlicer, it will detect that it has several shapes and ask the user if it should be treated as a multi-part object, then there are two options:

Yes) create individual extrusions for each shape making possible to change colors/heights/settings individually.

No) fuse the shapes and extrude them as a whole (this is how it works now).

I'm not sure at this point how strokes are now being dealt but I guess the right vector design should get rid of any stroke and just take in count the shapes (usually previewed as "fill").

Am I right?

GodricCZ, 2024-06-05:
SVG have Mask, ideal for this function is load SVG into Prusaslicer with Mask and split to singles Mask. For example in picture typical logo. If I load this SVG into PrusaSlicer, I have only square in PrusaSlicer. I can't delete any mask
This fuicntion can be best for Multicolor or other function. If I want have multicolor logo, I must generate separate SVG with separate Mask and import separate into PrusaSlicer.

[image]

pgilfernandez, 2024-06-06:
> SVG have Mask, ideal for this function is load SVG into Prusaslicer with Mask and split to singles Mask. For example in picture typical logo. If I load this SVG into PrusaSlicer, I have only square [...]

I don't really understand your point @GodricCZ, do you agree with me or you are adding new ideas? as far as I know, the Shell logo would render as a whole and you won't be able to get the yellow shape unless you create a separated SVG file with just that yellow shape...
Could you share PrusaSlicer screenshots to help me (us) understand you? thanks!

Jony01, 2024-06-06:
At this moment I suggest to add separated layers.

1. Separate layer to its own .svg file
Do not forget that for aligning SVG must have same size
(I put print mark rectangles into Top left and bottom right corner)
Note: Print marks are so small that are ignored in import but are calculated to size of layer.
[image]
2. Add first layer, scale and position it
[image]
3. Copy Paste volume --> select in right panel shell_red and press [ctrl]+[c] than [ctrl]+[v]
4. Change source svg of copied volume
[image]
[image]

source SVG:  [image]
separated layers with print marks:
[image]
[image]

pgilfernandez, 2024-06-07:
Hi @Jony01,

Now I understand your workaround, pretty nice.

BTW, a good solution for this multipart option would be to set up your SVG in **layers** and when you import it into PrusaSlicer it will prompt you to choose between a merged extrusion or a multipart extrusion driven by those layers you made in your SVG file.
That way they all would share the same origin and they would be placed exactly as you did in the SVG.

pgilfernandez, 2026-02-25:
After 2 years, the proposed feature is still relevant and super handy for graphic designers so, I finally took the job and did it, in this PR I include it all, I hope you like and second it: https://github.com/prusa3d/PrusaSlicer/pull/15265

pgilfernandez, 2026-09-03:
Does it make sense to turn this into a plugin for PrusaSlicer v3.0?