prusa3d/PrusaSlicer #11988: Drop Down menus appear outside of Prusaslicer and then disappear.

antietam76, 2023-12-28:
### Description of the bug

I was able to leave the drop-down menus open and leave Prusaslicer window, however when I did that the Drop-down menus remained on top of all windows. When I go back into Prusaslicer the drop-down menus disappear and I can no longer click on the menus to select an option. 

https://github.com/prusa3d/PrusaSlicer/assets/54750256/1ff68758-5cd3-4dd4-9708-aaca8cf194e9

### Project file & How to reproduce

I was able to reproduce it by:

1. opening the dropdown menu
2. Sliding cursor to outside of the window
3. moving cursor down to toolbar
4. opening another application (chrome etc.)
5. cycling through apps and watching drop-down staying on top of all windows
6. going back into Prusaslicer and then not being able to select from the affected drop down menu. 

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1 + win 64

### Operating system

Windows 11, 23H2

### Printer model

Prusa MK4

Jan-Soustruznik, 2024-01-11:
Hello @antietam76,

thank you for the report. We will try to fix it in one of the next releases.

checked created: SPE-2107

FidelCapo, 2024-01-29:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2024-01-29 by FidelCapo