prusa3d/PrusaSlicer #11989: No layers were detected bug

yarus23, 2023-12-28:
### Description of the bug

After splitting model to several submodels it cannot slice and show message 'No layers were detected'

### Project file & How to reproduce

[image]
[Epic Wolt bug.zip](https://github.com/prusa3d/PrusaSlicer/files/13789645/Epic.Wolt.bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 11

### Printer model

Ender 5 S1

neophyl, 2023-12-28:
The object tree list on the right shows that several of the split parts have errors (note the yellow triangles).  So why don't you do like the error message is telling you and fix the models ?

kubispe1, 2023-12-29:
Hi, @yarus23,
These two objects are likely causing the issue:
- Epic Wolt (1).stl_21
- Epic Wolt (1).stl_17 
For reference, could you provide the original model before it was split? Thanks

closed 2026-09-29 by prusaslicer-bot[bot]