prusa3d/PrusaSlicer #11990: Group print settings under corresponding filament profile

jw0785, 2023-12-29:
Currently, filament profiles inherit print or printer settings. it will be more intuitive to group print settings **under specific filament profiles.** 

For instance, tpu shouldn't share retraction setting with pla and other materials, this will guarantee a clog. This requires a new print setting, but the issue is that all those **filament specific** settings are mixed together relying solely on the naming scheme, which makes the management really messy.

neophyl, 2023-12-29:
Why haven’t you set the over ride values in the filament profile ?  That is what they are for after all.

jw0785, 2023-12-30:
that's one instance by chance that can be remedied with override, unfortunately also the only one. 

talk about speed, I print pla with 500mm/s, which cannot be overridden, and by no mean tpu can print that fast.

and even if there is an override, this **single override** will override **multiple print profile**, make 0.1 0.2 0.3 print the same speed, and some of the profile will over or under extrude.

sarusani, 2023-12-30:
You can override the speed at: Filament Settings->Advanced->Print Speed Override. It set's the Max volumetric speed for your filament, to prevent under extrusion on tall layer heights.

jw0785, 2024-06-10:
I just have an example at hand, where volumetric speed can't be the solution.
[image]
The wall prints just fine, however the top layer is absolutely horrible, as tpu cannot bridge as well, even it is within the range of volumetric speed.

So, should we then add another "max bridge speed " override? if this is not enough, another override? This really gets old quick. I still believe to avoid such dependency hell, the print setting should be filament specific, having all parameters needed rather than relying on one or two vague values.

closed 2026-09-29 by prusaslicer-bot[bot]