prusa3d/PrusaSlicer #11991: Filament selections do not appear on custom printer

TheTrueForce, 2023-12-29:
### Description of the bug

I have a Tronxy XY-2 Pro. Prusa slicer does not have a profile for this printer, so I created a custom printer for this. It's the only printer I have, so it's the only one I set up.

During setup, I selected Generic PLA and Generic PETG, and then when I went to set the filament, neither of those selections appeared. The only selection was named the same as the printer.

When I selected the "Add/Remove Filaments" option in that dropdown, it took me back to where I'd selected the filaments, and they were still selected. When I exited that dialog, they did not show up.
They also do not appear in the dropdown in the filament settings tab.

I would have expected those filaments to show up for selection.

[image]

### Project file & How to reproduce

Start with a clean install.
Set up a custom printer from scratch.
Set Custom PLA and Custom PETG as the filaments.
Attempt to set filament to one of those selected.

[Lithophane_Thickness_Calibration_Tile.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13790734/Lithophane_Thickness_Calibration_Tile.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1+linux-x64-GTK2

### Operating system

Arch Linux

### Printer model

Tronxy XY-2 Pro

Jan-Soustruznik, 2024-01-02:
Hello, @TheTrueForce, 

Thank you for your report. We will try to fix in some next release. 

issue created: SPE-2094

lukasmatena, 2024-02-09:
Fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2). Closing.

closed 2024-02-09 by lukasmatena