prusa3d/PrusaSlicer #11992: PrusaSlicer 2.7.1 fails to build on Linux with SLICER_GUI=OFF

rgmann, 2023-12-29:
### Description of the bug

I am attempting to build PrusaSlicer on Linux in a headless configuration. The build is failing during linking with the following error:

```
3935.9 [ 63%] Linking CXX static library liblibslic3r.a
3937.4 [ 63%] Built target libslic3r
3937.4 [ 63%] Building CXX object src/CMakeFiles/PrusaSlicer.dir/PrusaSlicer.cpp.o
3944.6 /PrusaSlicer/src/PrusaSlicer.cpp: In member function ‘int Slic3r::CLI::run(int, char**)’:
3944.6 /PrusaSlicer/src/PrusaSlicer.cpp:189:31: error: ‘Slic3r::GUI’ has not been declared
3944.6   189 |         if (std::find(Slic3r::GUI::OpenGLVersions::core_str.begin(), Slic3r::GUI::OpenGLVersions::core_str.end(), opengl_version_str) == Slic3r::GUI::OpenGLVersions::core_str.end()) {
3944.6       |                               ^~~
3944.6 /PrusaSlicer/src/PrusaSlicer.cpp:189:78: error: ‘Slic3r::GUI’ has not been declared
3944.6   189 |         if (std::find(Slic3r::GUI::OpenGLVersions::core_str.begin(), Slic3r::GUI::OpenGLVersions::core_str.end(), opengl_version_str) == Slic3r::GUI::OpenGLVersions::core_str.end()) {
3944.6       |                                                                              ^~~
3944.6 /PrusaSlicer/src/PrusaSlicer.cpp:189:146: error: ‘Slic3r::GUI’ has not been declared
3944.6   189 |         if (std::find(Slic3r::GUI::OpenGLVersions::core_str.begin(), Slic3r::GUI::OpenGLVersions::core_str.end(), opengl_version_str) == Slic3r::GUI::OpenGLVersions::core_str.end()) {
3944.6       |                                                                                                                                                  ^~~
3944.6 /PrusaSlicer/src/PrusaSlicer.cpp:190:35: error: ‘Slic3r::GUI’ has not been declared
3944.6   190 |             if (std::find(Slic3r::GUI::OpenGLVersions::precore_str.begin(), Slic3r::GUI::OpenGLVersions::precore_str.end(), opengl_version_str) == Slic3r::GUI::OpenGLVersions::precore_str.end()) {
3944.6       |                                   ^~~
3944.6 /PrusaSlicer/src/PrusaSlicer.cpp:190:85: error: ‘Slic3r::GUI’ has not been declared
3944.6   190 |             if (std::find(Slic3r::GUI::OpenGLVersions::precore_str.begin(), Slic3r::GUI::OpenGLVersions::precore_str.end(), opengl_version_str) == Slic3r::GUI::OpenGLVersions::precore_str.end()) {
3944.6       |                                                                                     ^~~
3944.6 /PrusaSlicer/src/PrusaSlicer.cpp:190:156: error: ‘Slic3r::GUI’ has not been declared
3944.6   190 |             if (std::find(Slic3r::GUI::OpenGLVersions::precore_str.begin(), Slic3r::GUI::OpenGLVersions::precore_str.end(), opengl_version_str) == Slic3r::GUI::OpenGLVersions::precore_str.end()) {
3944.6       |                                                                                                                                                            ^~~
3948.6 make[2]: *** [src/CMakeFiles/PrusaSlicer.dir/build.make:76: src/CMakeFiles/PrusaSlicer.dir/PrusaSlicer.cpp.o] Error 1
3948.6 make[1]: *** [CMakeFiles/Makefile2:614: src/CMakeFiles/PrusaSlicer.dir/all] Error 2
3948.6 make: *** [Makefile:146: all] Error 2
------
prusa-slic3r.Dockerfile:30
--------------------
  28 |     WORKDIR /PrusaSlicer/build
  29 |     RUN cmake .. -DSLIC3R_GUI=no -DSLIC3R_STATIC=1 -DSLIC3R_GTK=3 -DSLIC3R_PCH=OFF -DCMAKE_PREFIX_PATH=$(pwd)/../deps/build/destdir/usr/local
  30 | >>> RUN make
  31 |     
  32 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c make" did not complete successfully: exit code: 2
```

My build steps are as follows:
```
git clone --depth 1 --branch version_2.7.1 https://github.com/prusa3d/PrusaSlicer
cd ./PrusaSlicer/deps
mkdir build
cd build
cmake .. -DDEP_WX_GTK3=ON
make

cd ../../build
cmake .. -DSLIC3R_GUI=OFF -DSLIC3R_STATIC=ON -DSLIC3R_GTK=3 -DSLIC3R_PCH=OFF -DCMAKE_PREFIX_PATH=$(pwd)/../deps/build/destdir/usr/local
make -j4
```

Why is PrusaSlicer.cpp referencing `Slic3r::GUI` if I'm building in a headless configuration with `SLIC3R_GUI=OFF`? If `SLIC3R_GUI=OFF`, does `Slic3r::GUI` even get built?

### Project file & How to reproduce

A project file is not applicable for this bug report, since the error is happening during the build process. However, the following Dockerfile will enable you to exactly reproduce the issue.

```
FROM amazonlinux:2023 as build-image

RUN yum -y upgrade
RUN yum -y install gzip tar git perl wget cmake gtk3-devel dbus-devel mesa-libGLU-devel glibc-locale-source glibc-langpack-en glibc-common
RUN yum -y groupinstall "Development Tools"

RUN localedef -i en_US -f UTF-8 en_US.UTF-8
ENV LANG en_US.UTF-8  
ENV LANGUAGE en_US:en  
ENV LC_ALL en_US.UTF-8 

# Clone and build Slic3r
WORKDIR /
RUN git clone --depth 1 --branch version_2.7.1 https://github.com/prusa3d/PrusaSlicer
WORKDIR /PrusaSlicer/deps
RUN mkdir build
WORKDIR /PrusaSlicer/deps/build
RUN cmake .. -DDEP_WX_GTK3=ON
RUN make

WORKDIR /PrusaSlicer/build
RUN cmake .. -DSLIC3R_GUI=OFF -DSLIC3R_STATIC=ON -DSLIC3R_GTK=3 -DSLIC3R_PCH=OFF -DCMAKE_PREFIX_PATH=$(pwd)/../deps/build/destdir/usr/local
RUN make -j4
```

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Amazon Linux 2023

### Printer model

I3 MK3S+

rgmann, 2023-12-29:
This actually looks like a non-platform-specific issue that will impact any build using the `SLIC3R_GUI=OFF` flag. Perhaps that flag isn't used very often.

It looks like line 177-224 in PrusaSlic3r.cpp should be wrapped in a `#if SLIC3R_GUI .... #endif` block.

```
#if ENABLE_GL_CORE_PROFILE
    // search for special keys into command line parameters
    auto it = std::find(m_actions.begin(), m_actions.end(), "gcodeviewer");
    if (it != m_actions.end()) {
        start_gui = true;
        start_as_gcodeviewer = true;
        m_actions.erase(it);
    }

    it = std::find(m_actions.begin(), m_actions.end(), "opengl-version");
    if (it != m_actions.end()) {
        std::string opengl_version_str = m_config.opt_string("opengl-version");
        if (std::find(Slic3r::GUI::OpenGLVersions::core_str.begin(), Slic3r::GUI::OpenGLVersions::core_str.end(), opengl_version_str) == Slic3r::GUI::OpenGLVersions::core_str.end()) {
            if (std::find(Slic3r::GUI::OpenGLVersions::precore_str.begin(), Slic3r::GUI::OpenGLVersions::precore_str.end(), opengl_version_str) == Slic3r::GUI::OpenGLVersions::precore_str.end()) {
                boost::nowide::cerr << "Found invalid OpenGL version: " << opengl_version_str << std::endl;
                opengl_version_str.clear();
            }
        }

        if (!opengl_version_str.empty()) {
            std::vector<std::string> tokens;
            boost::split(tokens, opengl_version_str, boost::is_any_of("."), boost::token_compress_on);
            opengl_version.first = std::stoi(tokens[0].c_str());
            opengl_version.second = std::stoi(tokens[1].c_str());
        }
        start_gui = true;
        m_actions.erase(it);
    }

    it = std::find(m_actions.begin(), m_actions.end(), "opengl-debug");
    if (it != m_actions.end()) {
        start_gui = true;
#if ENABLE_OPENGL_DEBUG_OPTION
        opengl_debug = true;
#endif // ENABLE_OPENGL_DEBUG_OPTION
        m_actions.erase(it);
    }
#else
    // are we starting as gcodeviewer ?
    for (auto it = m_actions.begin(); it != m_actions.end(); ++it) {
        if (*it == "gcodeviewer") {
            start_gui = true;
            start_as_gcodeviewer = true;
            m_actions.erase(it);
            break;
        }
    }
#endif // ENABLE_GL_CORE_PROFILE
```

rgmann, 2023-12-29:
Another issue is that when `SLIC3R_GUI=OFF` and `SLIC3R_BUILD_TESTS=ON`, cmake still attempts to build tests that depend on `libslic3r_gui`.

lukasmatena, 2024-01-02:
Thanks, you are correct, this is a mistake. We will fix it.
SPE-2093

rgmann, 2024-01-03:
@lukasmatena Thank you!

lukasmatena, 2024-01-03:
@rgmann Just out of curiosity. CLI is available even when GUI is compiled. What is your use case for the GUI-less PrusaSlicer?

rgmann, 2024-01-09:
I can create a pull request for this if that would be acceptable.

lukasmatena, 2024-02-09:
Fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2). Closing.

closed 2024-02-09 by lukasmatena

mmiscool, 2024-02-10:
Quick question related to headless usage. Many years ago when I was using slic3r the package included Slic3r-console.

Is there any chance that a pre-built binary release for linux could be generated in a similar manner to the other the pre-built generic linux binaries like the "[PrusaSlicer-2.7.2-alpha2+linux-x64-GTK2-202402091359.tar.bz2](https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.7.2-alpha2/PrusaSlicer-2.7.2-alpha2+linux-x64-GTK2-202402091359.tar.bz2)" 

This would make it easier for installing in a docker container with out having to go through the whole build process.