prusa3d/PrusaSlicer #11995: Finishes work at the slicing stage

12288221, 2023-12-29:
### Description of the bug

Finishes work at the slicing stage.
[image]

### Project file & How to reproduce

[3DBenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/13795197/3DBenchy.zip)
Quit with "save project as"
Slice the model

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1 +win64

### Operating system

Windows 7

### Printer model

Elegoo Neptune 3 plus

12288221, 2023-12-29:
[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/13795227/PrusaSlicer.zip)

kubispe1, 2023-12-29:
Hi, could you please attach the .3mf file that is causing you trouble? I reconstructed the project from your files, but everything seems to be working fine for me.
[3DBenchy_kubispe1.zip](https://github.com/prusa3d/PrusaSlicer/files/13796184/3DBenchy_kubispe1.zip)

lukasmatena, 2024-02-18:
No response, closing.

closed 2024-02-18 by lukasmatena