prusa3d/PrusaSlicer #11999: 3mf file parse error

trivalik, 2023-12-30:
### Description of the bug

Import a 3mf file created from freecad 0.21.1 fails with:

```
Error (Invalid 3MF format) while parsing '3D/3dmodel.model' at line 2115
```
Part:
[FanDuctStd22-FanDuctStd.zip](https://github.com/prusa3d/PrusaSlicer/files/13798991/FanDuctStd22-FanDuctStd.zip)

### Project file & How to reproduce

`File` -> `Import` -> first entry for 3mf and select `FanDuctStd22-FanDuctStd.3mf`.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1 from dff602be63fb11bd6f566a4b54bb9423cc324d00

### Operating system

Windows 10

### Printer model

anet A6

Jony01, 2024-01-02:
Issue is made by using object type=`surface`, which is curently NOT supported.
The setting of transformation for uknown object caused an error during import.
I agree that message is missleading and will be fixed in future.

To fix your 3mf: Set object type to value `model` (in "FanDuctStd22-FanDuctStd.3mf\3D\3dmodel.model" on line 5)
[FanDuctStd22-FanDuctStd.zip](https://github.com/prusa3d/PrusaSlicer/files/13810407/FanDuctStd22-FanDuctStd.zip)

smirgol, 2024-07-10:
I'm having a similar issue with this model:
https://www.printables.com/model/932644-ariane-6/files

Opening it in Prusa Slicer 2.8.0 I get the following error:
```
[2024-07-10 11:17:53.625931] [0x000073bde11c6c80] [error]   Found component with invalid object id
[2024-07-10 11:17:53.625944] [0x000073bde11c6c80] [error]   Error (Invalid 3MF format) while parsing '3D/3dmodel.model' at line 34
[2024-07-10 11:17:53.625951] [0x000073bde11c6c80] [error]   Archive does not contain a valid model
```

The suggested fix doesn't work for me, as the object type already is `model`.
Looking at the file header, it looks like it was exported from the bambu slicer, if that helps. :-/

Jony01, 2024-07-10:
[smirgol](https://github.com/smirgol) 
FYI: 3mf use model with type=`"other"`, which is NOT allowed by  [3mf consortium](https://3mf.io/) to be inside of build. 

CITE:
> items MUST NOT reference objects of type "other", either directly or recursively.

[[link](https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md#3431-item-element)]

As you mentioned by error: it is found in '3D/3dmodel.model' at line 34, where is reference to `path="/3D/Objects/object_1.model" objectid="8"` 
in file '/3D/Objects/object_1.model' at line 1342495 is 
`   <object id="8" p:UUID="0001000e-81cb-4c03-9d28-80fed5dfa1dc" type="other">`

Shame on the Application that creates invalid 3mf.

Jony01, 2024-07-10:
Fixed: 
https://www.printables.com/model/938988-fixed-3mf

closed 2026-09-29 by prusaslicer-bot[bot]