prusa3d/PrusaSlicer #12000: 2.7.1 crashing every time i click slice.

Ireland-92, 2023-12-31:
### Description of the bug

It says "program not responding" and forces me to click close. Even on the most basic model. This is my first time ever using prusaslicer or a 3d printer in general. I have a very old laptop, but when looking up the system requirements, i just barely make the cut in all categories shown on prusa website. 

Old sony laptop with windows 7 64 bit. 4gb ram. 2.2ghz processors. 

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 7 64 bit

### Printer model

Prusa mini +

Ireland-92, 2023-12-31:
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	prusa-slicer.exe
  Application Version:	2.7.1.0
  Application Timestamp:	657872dd
  Fault Module Name:	ig4icd64.dll
  Fault Module Version:	8.15.10.1855
  Fault Module Timestamp:	4a6f79ac
  Exception Code:	c0000005
  Exception Offset:	000000000003ee1f
  OS Version:	6.1.7601.2.1.0.768.3
  Locale ID:	1033
  Additional Information 1:	b40c
  Additional Information 2:	b40c0642f63647c484d3306e56876349
  Additional Information 3:	f462
  Additional Information 4:	f462f6df8b8a9a242b630ed682d489f5

Read our privacy statement online:
  http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0409

If the online privacy statement is not available, please read our privacy statement offline:
  C:\Windows\system32\en-US\erofflps.txt

Ireland-92, 2023-12-31:
PrusaSlicer
Version:   2.7.1+win64
Build:     PrusaSlicer-2.7.1+win64-202312121425

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 7 (build 7601, Service Pack 1), 64-bit edition
Total RAM size [MB]: 4,126MB
OpenGL installation
GL version:   2.1.0
Profile:      Compatibility
Vendor:       Intel
Renderer:     Mobile Intel(R) 4 Series Express Chipset Family
GLSL version: 1.20.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_3DFX_texture_compression_FXT1
GL_ARB_color_buffer_float
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_fragment_program
GL_ARB_fragment_shader
GL_ARB_half_float_pixel
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_shader_objects
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_transpose_matrix
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_compiled_vertex_array
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_object
GL_EXT_multi_draw_arrays
GL_EXT_packed_pixels
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture3D
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_swizzle
GL_IBM_texture_mirrored_repeat
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_WIN_swap_hint
</details>

lukasmatena, 2024-01-02:
Thanks. Your graphics card is too old and only supports OpenGL 2.1. PrusaSlicer requires 3.2. The bug is that PrusaSlicer does not report an error and instead tries to continue. We will fix it.

You can run the application from the command line as `prusa-slicer.exe --sw-renderer`, which will force PrusaSlicer to use software rendering. It will be slow, but it should work.

SPE-2092

enricoturri1966, 2024-01-03:
@Ireland-92
Could you please download and test the following experimental build:
https://slicerbuilds.prusa3d.com/PrusaSlicer-2.7.1+2-et-spe2092-271-win64-gf6943f87d-202401031257.zip
?

Expected behavior: PrusaSlicer should now automatically select the software renderer instead of crashing.

Ireland-92, 2024-01-03:
Thank you for the response/help. Unfortunately this is my first time using
a 3d printer and prusaslicer etc. I was able to find the command prompt but
i typed that in and it did not work. I must have done something wrong.

Could you possibly give me brief step by step directions on how to do this?
Sorry for my lack of understanding on the process. Thank you very much for
your time and help.

P.S. i was able to get prusaslicer runninf on someone elses newer computer,
however i am hoping to get it to work on my personal laptop if possible.

Ireland-92, 2024-01-04:
I downloaded the file. The folder has about 15 things in it. Trying to
click and open the file prusa-slicer {application} gave me an error saying
that file MSVCP was missing. i then extracted all files from the folder.
Roughly 993 items were extracted. I then clicked and tried to open the
files again. Now it tells me "Prusa slicer was unable to automatically
switch to MESA OpenGL library. Please run with '--sw-renderer' option.

If i click openGL32 in the MESA folder, i select open with
prusaslicer setup. It then basically goes through installing
prusaslicer again. Then when i open it up and try to clice something, it
just says not responding and then closes like it has been.

Thanks for your help

enricoturri1966, 2024-01-04:
@Ireland-92

To run our applications with --sw-renderer option:

1) Download file:
https://slicerbuilds.prusa3d.com/PrusaSlicer-2.7.1+2-et-spe2092-271-win64-gf6943f87d-202401031257.zip
2) Unzip it on your hard disk
3) Download and copy into the newly created \PrusaSlicer-2.7.1+2-et-spe2092-271-win64-gf6943f87d-202401031257 folder the following batch files:
https://slicerbuilds.prusa3d.com/run_prusaslicer.bat
https://slicerbuilds.prusa3d.com/run_prusaslicer-console.bat
https://slicerbuilds.prusa3d.com/run_gcodeviewer.bat
4) Double clicking on any of those files (i.e. from File Explorer) will launch the correspondent executable with the --sw-renderer option

nealculiner, 2024-01-07:
New power computer built with ASUS ROG STRIX 4090, Intel i9-14900KF, no overclocking, but 4200 ram and I'm an experienced Prusa user and I'm also getting non stop crashing when slicing with 2.7.1. Extremely disappointed.

FidelCapo, 2024-02-01:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2024-02-01 by FidelCapo

Harvie, 2024-03-13:
I think this broke prusa slicer for me on thinkpad x201 with ArchLinux.
2.7.1 worked with opengl without any issues, 2.7.2 (after this fix) refuses to initialize 3d view.
Can you please add option to force OpenGL on older GPU? I can take the risk of slicer crashing, since i've never had any issues so far on this GPU...

Lorenzacosplay, 2024-12-01:
User with experience using Prusaslicer here - this error (unable to automatically switch to MESA OpenGL library) just popped up for me on 2.7 and persists after updating to the latest 2.8.1. @FidelCapo safe to say the issue is alive and kicking still :/