prusa3d/PrusaSlicer #12005: 2.7.1: One particulare .3mf project crashes when changing printer from XL to MK4

ssill2, 2024-01-01:
### Description of the bug

I have a 5T XL running latest firmware and using non-IS 0.6 XL profile.  I also have two MK4 printers.  I have one particular .3mf file that was originally created for the XL where I was going to use soluble supports.  I decided to print this part on one of my MK4s.  I simply open the .3mf in prusa slicer 2.7.1 and change the associated printer from the XL to MK4 and then select discard at the three prompts.   It crashes

### Project file & How to reproduce

Simply open this file.  Change the printer from XL 5T 0.6 non-IS to an MK4 non-IS.   Click discard on the three modal dialogs and it crashes.
[Ochi_Dagger_sliding_pointer.zip](https://github.com/prusa3d/PrusaSlicer/files/13804950/Ochi_Dagger_sliding_pointer.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 11

### Printer model

Prusa XL 5T 0.6 and Prusa MK4

lukasmatena, 2024-01-02:
I was unable to reproduce the crash. It may depend on what profiles you have installed. Can you please send us your full config bundle (`File->Export->Export Config Bundle`) ? Thanks.

ssill2, 2024-01-02:
yep,

[PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/13809503/PrusaSlicer_config_bundle.zip)

lukasmatena, 2024-01-02:
Thanks, we can reproduce the problem now. We will look into it.

ssill2, 2024-01-02:
> Thanks, we can reproduce the problem now. We will look into it.

awesome thanks!

lukasmatena, 2024-02-09:
Fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2). Closing.