prusa3d/PrusaSlicer #12006: First layer temperature in filament settings not obeyed

troth530, 2024-01-01:
### Description of the bug

First layer temperature in filament settings not obeyed.
This issue is probably usually masked when start gCode has one of the following lines.
M104 S[first_layer_temperature] or
M109 S[first_layer_temperature]

In this case I used start gCode of
M109 S212 with first_layer_temperature set to 260 as below
[image]

Here you see the first layer is only obeying start gCode setting intentionally set to other than filament setting to illustrate the bug.
[image]

Thank you!

### Project file & How to reproduce

[mobo-bottom-cover.zip](https://github.com/prusa3d/PrusaSlicer/files/13805072/mobo-bottom-cover.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1+win64

### Operating system

Windows 10 and Linux Mint 21

### Printer model

Default

rtyr, 2024-01-02:
Expected behavior. You have to either use the correct M104/109 format or not use M104/M109 commands at all.

[image]

closed 2024-01-02 by rtyr

troth530, 2024-01-02:
@rtyr Intuitively I would think that even heating of the bed and extruder are G-Code related and start G-Code comes before any and all heating and slicing including first layer.  But after reading it seems start G-Code is processed after initial heating, so I can see how temperature commands in start G-Code could conflict.   This has set an odd convention, which I suppose must remain at this point.

For now I can simply add a line such as M109 S[first_layer_temperature] to the end of my start gCode.