prusa3d/PrusaSlicer #12009: PrusaSlicer no longer works with FLSun QQ-S Pro

Danabler-42, 2024-01-02:
### Description of the bug

Prusa Slicer used to work fine with my FLSun QQ-S Pro Delta printer, in fact it was the only slicer I used that actually allowed me to get a proper print out of it. However after a recent update, no matter what I'm slicing, PrusaSlicer just returns a Gcode parsing error. I have re-installed Prusa Slicer, re-loaded my configurations, it even persisted across the most recent update, and on different PCs running PrusaSlicer. Unfortunately without PrusaSlicer I can't use this printer, as I've never been able to get anything resembling a decent print out of Cura. 

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusa Slicer 2.7.1

### Operating system

Windows 11 (but also 10 on a different machine)

### Printer model

FLSun QQ-S Pro

rtyr, 2024-01-02:
Fixed via configuration update 0.0.2.

Start g-code contained the variable `input_filename`, which is apparently no longer supported in PS2.7.0 (it may be a bug). I removed this variable, so it should work fine now.

neophyl, 2024-01-02:
Change it to Input_filename_base. It looks like the placeholder was updated. I’ve had this question pop up on the PS forum several times now.

closed 2024-01-02 by rtyr