prusa3d/PrusaSlicer #12013: Bug: Conflicts in G-Code paths have been detected at layer 0,00 mm.

Animo78, 2024-01-02:
### Description of the bug

When you auto arrange it sometimes puts parts which needed support to close together.
Giving an Error: Conflicts in G-Code paths have been detected at layer0, z=0.00 mm. Please reposition the conflicting objects .... further apart.

I thought this was resolved.

### Project file & How to reproduce

[XL Full bed for MK3S+.zip](https://github.com/prusa3d/PrusaSlicer/files/13813452/XL.Full.bed.for.MK3S%2B.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1 x64

### Operating system

Windows 11

### Printer model

XL Single Head

allanonmage, 2024-01-07:
I started working with the new version and had no idea what to do with this error.  Mine did not say to move them apart, but that's what solved it.

fordfrog, 2024-08-08:
[image]
i have a similar issue with 2.8.0. autoaligned items, overlap at layer zero.

srmalloy, 2025-08-24:
The problem still exists with 2.9.2; this is almost certainly due to the arrange function having no knowledge of the footprint of the supports/raft/brim for the objects; it just positions the objects on the platter as if they'll print naked. It would be a much more complex change to make the arrange function aware of support positions after a first attempt at slicing a platter, and adjust its arrangement based on the objects on the platter **and** their supports.

closed 2026-09-29 by prusaslicer-bot[bot]