prusa3d/PrusaSlicer #12014: Variable layer height prevents organic supports - even when part or layers are not supported.

itrmc, 2024-01-02:
### Description of the bug

When printing multiple items on a sheet, some need supports and some need variable layer height. Selecting organic supports is not allowed in this instance, although the organic-supported objects all have consistent layer heights.

Additionally, when supporting an object where only the top-few layers are variable, and layers far below these require organic supports, organic supports are not allowed.

This appears to be a simple check preventing use of organic supports. Enable a user-selectable override option if detection of these valid use-cases is too difficult.

### Project file & How to reproduce

[Valid-organic-variable layers.zip](https://github.com/prusa3d/PrusaSlicer/files/13814076/Valid-organic-variable.layers.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 11

### Printer model

All Prusa Models

KlausKraemer, 2024-01-07:
Just experienced something similar:

A project had supports from bottom to 12 mm. The remaining 79 mm were without support. Why shoudlnt it be possible to use variable layer heights above maximum support height?
[image]

smurfhunter, 2024-07-04:
This bug is still present on PrusaSlicer 2.8.0.

DJKyomi, 2024-07-11:
Same here on Linux Mint  Prusa Version 2.8.0 
Variable layer height is deactivated and i can't slice the object and all layers are 0.25

TerPuc, 2024-09-17:
Related #9462

closed 2026-09-29 by prusaslicer-bot[bot]