prusa3d/PrusaSlicer #12017: Snapmaker J1 Organic Supports

HuntNinja24, 2024-01-03:
### Description of the bug

On the organic supports it generates very strange from my other printers. Just wanted to send this out to see if this was normal or a known issue. 
[image]
 

### Project file & How to reproduce

[Snapmaker J1.zip](https://github.com/prusa3d/PrusaSlicer/files/13815535/Snapmaker.J1.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1+win64

### Operating system

Windows 11

### Printer model

Snampmaker J1  v2.6.2

rtyr, 2024-01-03:
It's caused by incorrect value (too high) for Branch diameter angle. I already notified the author of Snapmaker bundle (@macdylan) yesterday, see https://github.com/prusa3d/PrusaSlicer/pull/11798#issuecomment-1873940197.

I will wait a moment for his reaction/fix, otherwise I will fix it by myself.

rtyr, 2024-01-03:
Fixed via configuration update 1.1.1

closed 2024-01-03 by rtyr