prusa3d/PrusaSlicer #12019: prusaslicer 2.7.1 doesn't allow negative volume after a cut with connectors

m-cas, 2024-01-03:
### Description of the bug

After cutting a model (cut with separate pegs), it is still possible to add positive volumes but not to add negative ones.

### Project file & How to reproduce

just cut a model and see that the negative volume option is grayed out.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

macOS 

### Printer model

mk4

Jan-Soustruznik, 2024-01-05:
Hello @m-cas, 

could you provide more information(steps, video), please? I cannot reproduce on my PC with mentioned version of PrusaSlicer. Both options should be disabled "Add positive volumes" and "Add negative volume" when cut with connectors is made.

thank you, have a nice day

m-cas, 2024-01-05:
Hi @Jan-Soustruznik 
I have a video but it is 40mb and the limit seems to be 10!! Any other place where I can load it up?
I have loaded the 3mf file ... just take the part on the plate and cut with dowels.

Btw, why shouldn't I be able to add a volume (neg or pos) after a cut?

[all.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13842254/all.3mf.zip)

m-cas, 2024-01-05:
It looks like crap but at least you can see that the negative volume is greyed out and the positive one is not...
https://github.com/prusa3d/PrusaSlicer/assets/48383591/3b485251-c422-43bf-8344-4c1fcfecf114

TetsuoGreece, 2024-01-05:
I have the same problem. I created a 3d model of my brain from MRI scan and cut it. The negative volume is indeed greyed out!

TetsuoGreece, 2024-01-05:
I cut my model with Planar mode and it allows me to insert a negative volume! When I was using the dovetail it greyed out the negative volume. Hope this helps.

zpar-ky, 2024-01-05:
Related: #11157

Jan-Soustruznik, 2024-01-08:
Hello, @m-cas , 

That is expected behaviour, please check comment https://github.com/prusa3d/PrusaSlicer/issues/11157#issuecomment-1698708725

Anyway, If you want to modified cutted model, you can "invalidate cut info" (_r-click on cutted model -> invalidate cut info_), but after invalidate info, now can move with negative/positive volume added by cut feature.

I closed this issue as solved. 

Thank you, have a nice day

closed 2024-01-08 by Jan-Soustruznik