prusa3d/PrusaSlicer #12020: Travel moves are getting seperated into two travels

LidorBaum, 2024-01-03:
### Description of the bug

Im using Prusa XL at 5.1.2 and Prusa Slicer 2.7.1. when doing long travel moves, I see and hear that the move is seperated into two moves. when looking at the gcode, I can see it. its doing small move at F24007.498 and then a second move at F24000.
[image]

https://github.com/prusa3d/PrusaSlicer/assets/86967825/b69e5458-dffd-4f9d-b8d4-f4a6538c8321

### Project file & How to reproduce

[travel issue.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13821017/travel.issue.3mf.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

macOS 12

### Printer model

Prusa XL

BaconFase, 2024-01-03:
That's just the ramping lift that was added in 2.7.0 to help prevent stringing. When the ramping move has reached height it switches to regular horizontal movement.

Jan-Soustruznik, 2024-01-05:
Hello, @LidorBaum ,

as a mentioned by @BaconFase in comment above, it is a new feature ramping lift that will help to prevent stringing. See the changelog: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1

I closed this issue as solved.

thank you, have a nice day

closed 2024-01-05 by Jan-Soustruznik

LidorBaum, 2024-01-18:
Why is it keep happening even when ramping lift is off in the extruder settings under printing settings? Please have a look at the project file

Jan-Soustruznik, 2024-01-19:
Hello, @LidorBaum , try to check the _Filament Settings > Filament Overrides > Travel lift section_