prusa3d/PrusaSlicer #12022: auto-arrange tool to take into consideration clearance for supports

Russkil, 2024-01-04:
**Is your feature request related to a problem?
When using the autoarrange tool and then exporting gcode with the generate supports option switched on it errored to say they gcode had overlaps

**Describe the solution you'd like**
When using the auto arrange tool and having generate supports option on it arranges the parts further apart to allow for the placement of supports between models

**Describe how it would work**
When using the auto arrange tool and having generate supports option on it arranges the parts further apart to allow for the placement of supports between models

**Describe alternatives you've considered**
as a short term work arround if i run into this error i move the models further apart to overcome this.

**Additional context**
None I can think of

neophyl, 2024-01-04:
You can change the distance that auto arrange uses to separate the objects.  Right click on the Arrange tool.

rtyr, 2024-01-05:
I am closing this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3042.

closed 2024-01-05 by rtyr

Russkil, 2024-01-05:
> You can change the distance that auto arrange uses to separate the objects. Right click on the Arrange tool.

what would the suggested distance to prepredict supports?

neophyl, 2024-01-05:
Depends on what settings you have for the supports first layer expansion and if you use organic generally.  Organic trunks for higher parts can grow out a fair bit and combined with the first layer expansion can take up even more room.  Default first layer expansion is 3mm btw.  I don't use organic a lot (snug mainly) so minimum of 6mm is what I have set and generally I dont see the warning. ymmv