prusa3d/PrusaSlicer #12023: Dependencies fail to build on Windows when CPU has a large number of cores

tg73, 2024-01-04:
### Description of the bug

Following the [Windows build instructions](https://github.com/prusa3d/PrusaSlicer/blob/a061c508725aeb4dbccbe6d008212dcc4e78322f/doc/How%20to%20build%20-%20Windows.md) for building dependencies (the "manual" way), the `cmake --build .` step fails during OCCT compilation due to memory exhaustion.

My build machine has has 32 logical cores (13th Gen Intel(R) Core(TM) i9-13900K, 3000 Mhz, 24 Core(s), 32 Logical Processor(s)) and 64GB of RAM. When it gets to OCCT, the build process spawns 700+ `cl.exe` processes and exhausts system memory (all 64 gig of it).

[image]

It appears that the default parallelization strategy of cmake/msbuld/cl causes this crazy process storm, and it boils down to occurrences of `-m:32` appearing in the generated `.vcxproj` files. As cmake newbie, and despite much googling, I could not work out how to stop the vcx generator from emitting those `-m:32`. In the end I just did a search and replace in the `.vcxproj` files, replacing `-m:32` with `-m:1`. And in `deps\build\CMakeCache.txt` I added `/MP1` to `CMAKE_CXX_FLAGS` and `CMAKE_C_FLAGS`.

It would be good if a cmake guru could fix this in the cmake code or update the build instructions for windows to cover this scenario.

### Project file & How to reproduce

See above.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

refs/tags/version_2.7.1

### Operating system

Windows 11 x64

### Printer model

N/A

jmerdich, 2024-01-28:
Also encountered here. Note that additionally, MVSC will not cancel the sub-build of each dependency, making this extremely painful. Microsoft is apparently working on a fix in msbuild (though that doesn't save us from trying to spawn hundreds of processes in the first place).

From a dive into the code, this behavior can be altered by the cmake variable `DEP_MAX_THREADS` (eg. `cmake -DDEP_MAX_THREADS=1 ..`), as used by AddCMakeProject.cmake. Note this controls the number of jobs in *each sub-build* and not the main build, but trying to limit the main build's threads with `-j1` still made me run out of memory. Setting `DEP_MAX_THREADS` successfully got me from overwhelming my "measly" 48GB of RAM (2GB/core) to actually building. I'm not sure if this is the canon way to build, but it gets me away from build system pain and onto real work.

Thanks goes to @tamasmeszaros who added that knob a few months ago!

strepto42, 2025-02-02:
Happening here with an AMD 7950X / 32 logical processors also.

closed 2026-09-29 by prusaslicer-bot[bot]