prusa3d/PrusaSlicer #12024: Prusa Mini has no way to turn off Binary Gcode aka bgcode

desktopbob, 2024-01-04:
### Description of the bug

Prusa Mini has no way to turn off Binary Gcode aka bgcode.

[image]

### Project file & How to reproduce

Go to print settings
Output options
Nothing there to change as there was supposed to be

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusa Slicer 2.7.1

### Operating system

Windows 10

### Printer model

Prusa Mini

desktopbob, 2024-01-04:
the last update fixed the mk3 but not the mini

mix579, 2024-01-04:
You can turn it off under Printer Settings→General→Firmware→Supports binary gcode

Or turn it off for all printers in Preferences→Other→Use binary gcode etc

desktopbob, 2024-01-05:
Perfect thanks !

closed 2024-01-05 by rtyr