prusa3d/PrusaSlicer #12025: Program crash (reliably) with Layer height 2.5, but not 2.0 or 2.8 or 3.0

burnbasket, 2024-01-04:
### Description of the bug

The model included in this report will crash version 2.6.1 of PrusaSlicer from the Debian Sid repo every time gcode generation is attempted. It happens on "File->Export->Export as G-code" as well as on the preview button on the "Plater" tab.

This is specific to the setting of  a "Layer height" value of  "0,25". Everything works fine with different values, e.g. "0,2", "0,3" or "0,28".

The "First layer height" value was always set to "2,0" while this occurred.

### Project file & How to reproduce

[Winder_left_hand_frame.zip](https://github.com/prusa3d/PrusaSlicer/files/13835825/Winder_left_hand_frame.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Debian GNU/Linux trixie/sid

### Printer model

Ender 3

lukasmatena, 2024-01-05:
Reproduced in 2.6.1, not reproduced in 2.7.1. It seems to be a duplicate of something that we already fixed, most likely #11267. Closing.

closed 2024-01-05 by lukasmatena

ganzgustav22, 2024-01-16:
You haven't fixed anything it keeps crashing on me in different variants:

1. Model preview screeen turns white, I can still save project but cannot do anything with the model anymore.
2. The app crashes with the usual windows popup telling me the app crashed
3. Not a real crash but the Slicer makes the WHOLE MACHINE become unresponsive for like 30 seconds or so, mouse cursor does not move anymore and also CTRL-ALT-DELETE doesn't do anything.
4.