prusa3d/PrusaSlicer #12028: Minimum shell thickness

laptop-laptop, 2024-01-05:
**Is your feature request related to a problem? Please describe.**

In 2.7.1 and pervious versions, under Print Settings->Layers and perimeters, Horizontal shells, there is a "Minimum shell thickness" Top and Bottom fields. Can you please add an option for automatic calculation of these two thickness based on corresponding Top and Bottom values under "Solid layerss"?

I read the descriptions but I don't understand what they mean and how to calculate those two values. So far, in different projects,
I just change the Top and Bottom fields under "Solid layers" while keeping the corresponding values under "Minimum shell thickness"
unchanged. Am I supposed to change these values whenever I change the number of Top and Bottom Solid Layers? 
What are the recommended values?

**Describe the solution you'd like**
Please see above

**Describe how it would work**
Please see above

neophyl, 2024-01-05:
Minimum shell thickness is exactly what its described as.  Try reading the tooltips.  
If you set it to for example 1mm for Minimum shell thickness top then even if you set top layers to 1 you will get 1mm thickness of top layers, so with a 0.2mm layer height you will get 5 layers.  If you change to a 0.1mm layer height you will get 10 layers.  It will make sure your top is always a MINIMUM thickness to match the minimum shell thickness setting.

The only time you won't is when the top layers is set to Zero which greys out the setting anyway so should be obvious that its not applicable.

This works the same with the bottom layers too.

WizardUli, 2024-01-27:
I don't understand why it is possible to use both definition at the same time when only the higher minimum takes place. That's confusing UI.

Let's consider layer height 0.2mm
- If I define minimum of 5 bottom layers and minimal bottom shell of 0.6mm then the minimum it's going to always be 1mm (5 layers).
- If I define minimal bottom shell of 1mm and minimal 3 bottom layers then it's going to always be 1mm (5 layers).

It could very well have been just one value with possibility to include or not include unit `mm` at the end.

neophyl, 2024-01-27:
You forget though that there are tools like variable layer heights or modifiers. It’s quite possible to use one of them and all of a sudden your configured 5 layers is now 5 layers of 0.04mm, that would give you a very thin top or bottom. By specifying a minimum thickness it guards against such problems. 

Just because you find it confusing doesn’t mean every one else does.  
if you think the UI is confusing here just wait till you discover other parts of it. Have fun learning it’s idiosyncrasies.

WizardUli, 2024-01-28:
Variable layer height explains it 100% then. Haven't thought about it. Thank you.

closed 2026-09-29 by prusaslicer-bot[bot]