prusa3d/PrusaSlicer #12029: First Support Interface Layer not recognized as bridging

Mike-Flexicon, 2024-01-05:
### Description of the bug

I am printing on an IDEX FDM machine (Modix Big-60) using ABS with HIPS as the supports.

Using honeycomb snug supports with 4 interface layers.

On the first interface layer, where the hips must bridge across the supports, I get really bad sagging/failed bridging without the part cooling fan running.  If I manually turn the fan on during the first interface layers, I get proper bridging.  It is clear in the slicer that the first interface layer is not being detected as bridging, thus no fan is running.  I could post process and manually add a fan on and fan off line in the code for every interface line, but it is only necessary on the first layer of the interface and might even be a detriment if ran on all interface layers, especially the final.

### Project file & How to reproduce

[TFC-00460057.zip](https://github.com/prusa3d/PrusaSlicer/files/13845444/TFC-00460057.zip)
[image]

I believe the issue is explained well above and you need only look through the project file to see what is what.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.7.1

### Operating system

Windows 11 Business 22H2

### Printer model

Modix Big-60 IDEX

neophyl, 2024-01-05:
You have contact distance configured as zero.  That means soluble supports as the setting label says.  What you probably don't realise is that when set to zero it turns OFF **all** bridge detection.  After all you are using soluble supports with zero distance so bridging isn't needed.

You can set a very small non zero value such as 0.01 for the contact distance to enable bridging but still put supports close to them.

Mike-Flexicon, 2024-01-05:
> You have contact distance configured as zero. That means soluble supports as the setting label says. What you probably don't realise is that when set to zero it turns OFF **all** bridge detection. [...]

That would be the case if I needed the fan on during the Last layer of the Support interface. But I need the fan on the 1st layer of the interface, which is the same material as the supports itself. I believe my screenshot clarifies that situation.

Thank you for the response!

the-ress, 2024-04-21:
> You have contact distance configured as zero. That means soluble supports as the setting label says. What you probably don't realise is that when set to zero it turns OFF **all** bridge detection. [...]

The problem with that is it'll disable the wipe tower.

[image]

And if you use something like paint-on support blockers, the bridges might be actually needed.

closed 2026-09-29 by prusaslicer-bot[bot]