prusa3d/PrusaSlicer #12033: Slicer - no layers for significant part of the model

viesturz, 2024-01-06:
### Description of the bug

See the attached project file.
When I slice it, there is a big part of the model that is not printed at all. A part on top of it is printed in thin air.

### Project file & How to reproduce

See the attached project.
[Dragon_Plate.zip](https://github.com/prusa3d/PrusaSlicer/files/13849131/Dragon_Plate.zip)

[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Win11

### Printer model

voron 2.4

viesturz, 2024-01-06:
Looking closer, the step file is missing some surfaces. 
Can PrusaSlicer detect this and warn?
[image]

viesturz, 2024-01-06:
Note that Fusion360 is able to open the STL correctly, so this is a STEP parsing bug on Prusa side.

mix579, 2024-01-06:
It does warn you. There are warning triangles next to the part names, and when you click on them it shows you that both parts have a large number of open edges. When you load it into Fusion and go to the Mesh workspace and click on Repair, it also shows you that the part (I only looked at the main part) has open boundaries but it can repair but I don't know if the result is what you would expect

viesturz, 2024-01-07:
Thanks. Probably worth showing that somewhat more prominently. That small exclamation mark is lost in the UI.
Anyhow seems to be working as intended, closing.

closed 2024-01-07 by viesturz