prusa3d/PrusaSlicer #12035: gaps in monotonic top/bottom fill

jmerc77, 2024-01-07:
### Description of the bug

these gaps...

[image]

...kind of annoying. happens on both the bottom and the top. they don't exits inside the part on 100% infill though.

### Project file & How to reproduce

[gaps.zip](https://github.com/prusa3d/PrusaSlicer/files/13852842/gaps.zip)

mainly use monotonic filled tops/bottoms.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1+Win64

### Operating system

windows 10

### Printer model

CR-10S (but heavily moded)

MichaelJLew, 2024-01-07:
Are they visible in the print?

BrienKing, 2024-01-07:
Yes, I have this issue too.   Prior to 2.7.1 the print was perfect.  Now I see little spots.  I'll get some "before" and "after" photos to post here.

jmerc77, 2024-01-07:
> Are they visible in the print?

yes, but barely. you have to look closely. especially with small extrusion widths or small nozzles. also, as a side note, as small as these parts are, the bad adhesion warnings do not appear. I'm ok with that though, because these were intentionally small as i was testing an openscad lib for some projects that is intended to be small.

BrienKing, 2024-01-24:
This is the before:
[image]

This is the after:
[image]

closed 2026-09-29 by prusaslicer-bot[bot]