prusa3d/PrusaSlicer #12037: Layer is divided

aclucca, 2024-01-07:
### Description of the bug

Why is in Layer 6 (1,2mm) splitted?
This results an inhomogenius top layer . . .

### Project file & How to reproduce

[Desktop.zip](https://github.com/prusa3d/PrusaSlicer/files/13854407/Desktop.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 10 64bit

### Printer model

PRUSA MK4

neophyl, 2024-01-07:
Correct and also expected operation for PS.  The red section is topfill so printed as topfill, the purple section is normal rectangular solid infill and so printed as such.

There are several issues where this has been reported previously where you could have found the answer to your question. 
For example https://github.com/prusa3d/PrusaSlicer/issues/8818 or https://github.com/prusa3d/PrusaSlicer/issues/9133

aclucca, 2024-01-07:
Ok, thank you for the explantion.

But why it isn´t printed as (one) bottom Layer?
I´ve choosen 28 Bottom Layers (also for adding weight) and in my opinion Layer 6 should also be one homogenius bottom layer.

neophyl, 2024-01-08:
Why would it be printed as a bottom layer when its the top most visible layer ?  You could configure 1000 bottom layers but if a layer is visible on top then it is classed as a top layer.

neophyl, 2024-01-08:
No it wont.  Its fundamental, part of that layer is visible and so gets printed as top layers, the part under the rest of the object is not visible and so sis not a top layer.  The current software will split that layer between the types as that is what it is designed to do.  It should make no different to your print as the 'join' where the types meet is under other layers printed on top anyway.

If I really really had to have that layer to be unbroken I would use a different slicer that has controls and settings allowing you to tweak such things.

closed 2026-09-29 by prusaslicer-bot[bot]