prusa3d/PrusaSlicer #12038: Normal Redo Shortcut

pravdomil, 2024-01-08:
can we have Redo to have shortcut Cmd + Shift + Z? like other software
 
[image]

hkienle, 2024-01-27:
Ctrl-Y is (also) a reasonable choice, at least on Linux.

samyk, 2024-06-15:
I agree the functions should be the system default for the OS.

On Linux, that's ctrl+Y and ctrl+shift+Z.
On macOS, that's cmd+shift+Z.
On Windows, that's ctrl+Y.

Confusingly on macOS if you attempt cmd+shift+Z, it ends up _undo'ing_ rather than redoing -- I just lost some work running into this nuance.

Editor for shortcuts would be nice but is more work and while editor is nice, system defaults should still prevail by default for accessibility and is less effort to implement as it doesn't make sense for all users of an OS to need to adjust commonly known shortcuts.

- macOS shortcuts: https://support.apple.com/en-us/102650
- Windows shortcuts (under "Copy, paste, and other general shortcuts"): https://support.microsoft.com/en-us/windows/keyboard-shortcuts-in-windows-dcc61a57-8ff0-cffe-9796-cb9706c75eec

hkienle, 2026-09-02:
Don't know what shortcut is used in 3.0, but it would be a great opportunity to finally have a consistent shortcut!

ex-nerd, 2026-09-02:
> Don't know what shortcut is used in 3.0, but it would be a great opportunity to finally have a consistent shortcut!

It's still ⌘Y. Really overdue fixing it to ⌘⇧Z. Even worse, it's not an actual menu option so you can't just go in and re-map it manually via System Settings:

[image]