prusa3d/PrusaSlicer #12039: CAD to Slicer friction

pravdomil, 2024-01-08:
**Fusion 360**
[image]

**PrusaSlicer alert 1**
[image]

**PrusaSlicer alert 2**
[image]

**PrusaSlicer place it with incorrect orientation**
[image]

Can please we have 0 alerts and geometry that matches CAD software? That would be really great!!!

neophyl, 2024-01-08:
Looking at your fusion image you have laid your objects out with increasing z height, (the blue axis), so PS is using your cad software orientation already.  

And of course it asks each time you load something , its not a global setting and will depend on what you are loading in at any particular time.

pravdomil, 2024-01-08:
[image]

neophyl, 2024-01-08:
Exactly, your image shows my point.  The z axis is blue and is up/down with respect to the platter.  Your 3 items are placed with the smallest being the lowest height above the platter and then they move further away.  SO PS is loading yor objects in at exactly the same orientation as they are placed in Fusion.

pravdomil, 2024-01-09:
https://github.com/prusa3d/PrusaSlicer/issues/12043

closed 2024-01-09 by pravdomil