prusa3d/PrusaSlicer #12040: Support pre-color mapped 3D models for import

rcmaniac25, 2024-01-08:
**Is your feature request related to a problem? Please describe.**
Various 3D model formats allow setting colors to vertices and more advanced software allow UV mapping of textures, but upon import, we do not get any of the color information.

**Describe the solution you'd like**
If color information is present in the model file, offer to include it for a multi-color print. If UV maps exist, apply or remesh the colors to the vertices in the model. This way the model retains the color information that was implemented in the modeling software.

**Describe how it would work**
Depending on how PrusaSlicer implements it, it could be as easy as assigning a color to a triangle or as advanced as remeshing the model so the colors can be applied. One case that may require additional UI, mapping a potentially large color set to the smaller number of filament options would be needed and would benefit from a visual of the remapped colors before actually trying to apply/remesh them.

**Describe alternatives you've considered**
Painstakingly applying painting details by hand.

**Additional context**
My initial experience comes from UV mapping and painting models and wishing something existed for multi-color prints, but the closest I saw to this came from the web slicer used by the Mosaic Palette and their stamping feature, where you would import an image, map the colors of the image, point the camera at the model and then apply. It would then apply the color to the model without hand painting. I didn't realize I have not made this request until I saw someone talking about a similar request and a script someone made to do this outside of a slicer: https://github.com/bambulab/BambuStudio/issues/1892

My best guess is that, sans the color mapping of the script, it can probably function the same on 3MFs from PrusaSlicer.

neophyl, 2024-01-09:
I think this one https://github.com/prusa3d/PrusaSlicer/issues/4503  is pretty much requesting the same thing.

rcmaniac25, 2024-01-09:
> I think this one #4503 is pretty much requesting the same thing.

The original inspiration was from dealing with OBJ in modeling software like Maya, the reminder (the ticket mentioned above) was also taking from OBJ.

Formats from PrusaSlicer that support color and/or texture:
- OBJ
- 3MF
- AMF
- STEP/STP

I'm not sure if anyone is using AMF, and STEP/STP is processed by a library so not sure if it deals with color. PrusaSlicer has it's own way of supporting colors in 3MF and if they want to support more options that would be great. But it probably means that this request is really to support color/UV information from OBJ.

SupItsSaige, 2024-05-16:
THIS!!!! PLEASEEE!!!! Especially with the Prusa XL finally making its way out, how is there not a more streamlined process for bringing in multi-color/material models? I get that a lot of people have been use to needing to manually separate the colors through the mesh itself as separate parts, but that is not efficient and definitely not possible for some of the more complex models out there.

Being able to keep the mesh color metadata from the 3D-modeling software would make the process so much more efficient and tuning a model would be so much easier. Is this currently being worked on? It is pretty crazy that such a useful similar request that was made 4 years ago still has not been implemented. This is absolutely a must have for those of us that are using Prusa XLs and MMUs for business purposes and have a lot of iterations.... Otherwise we are just making the process unnecessarily more difficult by keeping things status quo (manually breaking apart the meshes).

kulberda, 2024-05-25:
I just recently got an XL and had assumed the option to bring in a UV mapped OBJ with an image map for color information was one of the options to print multicolor.  I was surprised to find that it was not.  The ability to paint materials in the slicer itself is a start but not nearly as robust as Blender or ZBrush painting (or even manual UV map painting in Photoshop).

FEMit160, 2024-07-11:
Yes! Please add this feature to Prusa Slicer.

Bambu Studio now has this feature available as of the April 2024 1.9.1 release, so it should be doable in Prusa Slicer.

[https://wiki.bambulab.com/en/software/bambu-studio/release/release-note-1-9-1](url)

drackbolt, 2024-09-26:
Once I realized this feature wasn't yet available, I held off on buying the XL.  Wouldn't see any point to it without this ability!  Would be awesome to add it to Prusaslicer in general to keep up with the other fork updates.

gottesgerl, 2025-12-15:
+1 here as well. Have a 5-toolhead XL and would really appreciate this feature. Even more for everybody with the new INDX System.

agoode, 2026-09-05:
This is still a good feature request, see previous comments.