prusa3d/PrusaSlicer #12041: Slicer 2.7.1. Black build screen and crash on startup

asignbefore, 2024-01-09:
### Description of the bug

I just updated my slicer to 2.7.1 with the drivers. Upon installing, the program asked if I wanted to send my computer info to Prusa. After clicking anything, the program crashes. While this is happening, the build screen (originally OK looking) turns black. I tried to down-rev to 2.7.0 with the same results.

Computer info:
Processor	AMD Ryzen 9 4900H with Radeon Graphics            3.30 GHz
Installed RAM	16.0 GB (15.4 GB usable)

[image]

### Project file & How to reproduce

1. open software
2. wait about 3 seconds
3. it freezes and the build screen turns black

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0 

### Operating system

Windows 11 Home

### Printer model

MK4

asignbefore, 2024-01-10:
figured it out... I looked under Display Adapters under Device Manager and found there were two installed for my AMD Radeon graphics card.. removed one of them and it started working again without crashing

closed 2024-01-10 by asignbefore

Rino91, 2024-12-03:
We had the same problems on a few computers in the company after updating to W11 (Laptops with AMD APUs). The solution that worked for us was adding a slicer to "games" in the AMD Adrenaline control panel and disabling optimization on it. The same situation was with a few other programs, such as SolidWorks.