prusa3d/PrusaSlicer #12042: Since 2.6.0 it keeps crashing

ganzgustav22, 2024-01-09:
### Description of the bug

It keeps crashing. Either becomes completely unresponsive and I have to kill it. Or the build area becomes white and atleast I can still save the project and re-start it.

Also, it often makes my whole machine freeze for 10 seconds or more, i.e. mouse pointer is frozen, nothing can be done, even CTRL-ALT-DEL won't react.

Now I just encountered another crash, in vase mode it is stuck at 90% generating G-Code for 10 minutes or so (simple model).

Please, it was such a great slicer until 2.6.0. Now this :(

I did not change anything on my machine, just updated Prusaslicer.

### Project file & How to reproduce

Just use it.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-2.7.0

### Operating system

Windows 10

### Printer model

irrelevant

ganzgustav22, 2024-01-17:
[image]

NielschB, 2024-01-19:
Also here 2.7.1 seems to crash a lot. Have not been able to pinpoint the exact scenarios. It almost looks random however it seems that it happens especially when PrusaSlicer has been idling in the background for a while. Sometimes just even moving a part on the "bed" makes it crash

kubispe1, 2024-01-30:
Hi @NielschB and @ganzgustav22, this is a strange behavior. To resolve this issue, we need to reproduce it. Could you please provide me with the Help tab/System information? It might be a duplicate or similar to #12049, but I'm not sure.

NielschB, 2024-01-30:
PrusaSlicer
Version:   2.7.1+win64
Build:     PrusaSlicer-2.7.1+win64-202312121425

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (build 22621), 64-bit edition
Total RAM size [MB]: 17,143MB
OpenGL installation
GL version:   4.6.0
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1060 6GB/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
[... 313 lines cut ...]
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

closed 2026-09-29 by prusaslicer-bot[bot]