prusa3d/PrusaSlicer #12045: Permission denied when flashing firmware

ionv3n0m, 2024-01-10:
### Description of the bug

When trying to flash firmware to MK3/MK3s, it shows in the logging immediately after a permission denied message to the port on the printer. 

### Project file & How to reproduce

Attempted flash via Debian 12 laptop, with Prusaslicer-2.7.1+linux-x64-GTK3-202312121425.Appimage

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Prusaslicer-2.7.1+linux-x64-GTK3-202312121425

### Operating system

Debian 12

### Printer model

MK3 and MK3s

lukasmatena, 2024-01-10:
"Permission denied" might indicate that... well, that you don't have non-root access permission to the USD port. See e.g. #1055, #2640.

ionv3n0m, 2024-01-10:
I get the whole permissions bit, and not having it. What I almost expected, would've been a password confirmation box (like updating applications through UI, or running some other applications that need a temp root/sudo access).  

I'll go through the 2 examples and give one a go later, to report back here.  Just out of arguement's sake, is this documented in FAQs/KB/Prusa documentation?

SizzinSeal, 2024-01-13:
permission errors may also indicate a damaged USB port or cable. I've had it happen to me a few times, not on Prusa machines but with others. Try a different cable and USB port.

brtmax, 2025-01-22:
For anyone else having similar issues: 
If PrusaSlicer was installed via snap, you need to explicitly allow it to access USB devices: 
`sudo snap connect prusa-slicer:hardware-observe
sudo snap connect prusa-slicer:raw-usb
sudo snap connect prusa-slicer:removable-media
`

since snap packages run in a confined sandbox environment.

closed 2026-09-29 by prusaslicer-bot[bot]