prusa3d/PrusaSlicer #12049: PrusaSlicer crashes GPU while slicing simple model

ogsv, 2024-01-11:
### Description of the bug

When slicing the extremely simple model in attachment, PrusaSlicer 2.7.1 becomes unresponsive at different points in the process, sometimes at "generating perimeters", and sometimes at "searching for supports" (even though supports are deactivated).
Sometimes I can slice the part successfully once, but if I change some parameters and try to slice again, it always crashes.
First the aspect ratio of my display becomes completely messed up, then Fusion 360 crashes, then I get a popup telling me that my AMD graphics driver has crashed. I then have to reboot the system to get the display to recover.
AMD Ryzen 3 CPU, old Radeon RX550 GPU, 24 GB RAM, Windows 11, latest PRO drivers from AMD.
(Unfortunately, due to the display becoming almost useless after this crash, I have not been able to capture screenshots)

I can slice other models (usually more complex ones) without any issue, and I can slice this model in SuperSlicer_2.5.59 with no issues.

### Project file & How to reproduce

[Cartridge storage v2.zip](https://github.com/prusa3d/PrusaSlicer/files/13901671/Cartridge.storage.v2.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 11 build 23612

### Printer model

Hypercube Evolution

kubispe1, 2024-01-30:
Hi @ogsv, could you please provide me with the Help tab/System information?
Thanks

ogsv, 2024-01-31:
Hi @kubispe1 , sorry for the delay - here is the full sysinfo:
```
PrusaSlicer
Version:   2.7.1+win64
Build:     PrusaSlicer-2.7.1+win64-202312121425

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (build 23620), 64-bit edition
Total RAM size [MB]: 25,700MB
OpenGL installation
GL version:   4.6.0
Profile:      Core
Vendor:       ATI Technologies Inc.
Renderer:     Radeon RX550/550 Series
GLSL version: 4.60.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_AMD_blend_minmax_factor
GL_AMD_conservative_depth
GL_AMD_depth_clamp_separate
GL_AMD_gpu_shader_half_float
GL_AMD_gpu_shader_int16
GL_AMD_gpu_shader_int64
GL_AMD_multi_draw_indirect
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_sample_positions
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_trinary_minmax
GL_AMD_texture_cube_map_array
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
[... 226 lines cut ...]
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_timeline_semaphore
GL_OES_EGL_image
GL_OVR_multiview
GL_OVR_multiview2
GL_OVR_multiview_multisampled_render_to_texture
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
WGL_EXT_swap_control
</details>

```

closed 2026-09-29 by prusaslicer-bot[bot]