prusa3d/PrusaSlicer #12050: Prusa XL / MMU define Spool Join in PrusaSlicer

juergen852, 2024-01-11:
Printers having multiple extruders or MMUs can do spool join.

I would like to be able to define in the slicer which spools/printheads should be used for joining printmaterial and soluble material.

vk7krj, 2024-03-12:
I would like to add my support to this request- having a 5 head XL, being able to add spool-join to the job from the slicer would be a great help for me.

plucker, 2024-03-14:
this is essential please

Cap1a1n, 2024-03-22:
This is really needed

mkubista, 2024-03-24:
I would like this feature too. I know, it can be done on the printer when selecting tools for the new print. But it would be nice to do it in slicer, when im sending print by PrusaLink/Connect.

MontgomeryLC, 2024-03-30:
Yes please.

ex-nerd, 2024-03-30:
Would be nice to get it in the firmware when you run out of filament, too. I ran into this with a print yesterday but the filament I wanted to use was already loaded onto a different extruder. Had to cut the filament and re-load it onto the "primary."

vk7krj, 2024-03-30:
It would indeed be nice ex-nerd. My XL is in my workshop in a heated enclosure I built- and it's a real pain being able to remotely turn the printer on, then send the file over the network to the printer- then having to go out there on a cold wet night, turn off all the security, unlock everything, set up a spool join, then close up and reset everything- when it could so easily be done within the slicer in the comfort of my radio room and I can just watch the results on the camera.

mjrko84, 2024-05-02:
Absolutely essential! Please add this feature!

mwdaley, 2024-07-01:
It's great to see this in the firmware, but it really belongs in the slicer, especially when you know your print will take more filament than you have on one spool.  That could enable multiple partial spool uses (or a >1kg print) without having to babysit a filament runout.

nhustak, 2024-07-01:
I'm with you on it being in the slicer but you CAN go do it on the screen
so there's no reason to babysit your filament.
We're talking convenience here - and I'm all for it 8)

buchnerb, 2024-07-13:
If you can't add this to slicer where it belongs please add it to Connect. If I send a print to to connect through Slicer I don't get to add spool join at the printer it just starts.

FishyF1sh, 2024-08-04:
+1, please make it possible to configure spool join remotely.

jhoblitt, 2024-08-12:
As this feature is missing, it practically means the XL can only print from USB.  As tool remapping / spool joining is non-persistent from the console,, it means manual setup for every print.

Merlerault, 2024-09-04:
+1 my computer is not close to my XL and the time to go on my XL, I forget to modify the parameter before to launch the print on the printer XL's menu. Thank you

mholtmanns, 2024-09-19:
+1 Yes please!

yamez25, 2024-10-18:
We need spool join without printing from USB. This likely means spool join setup either needs to be persistent or G-code from the slicer (or preferably both). Spool join is not usable for us in the current state.

alvomane, 2024-11-12:
I would also like to add my support to this request!

devicenul, 2024-12-03:
+1 - I'd very much like this and the ability to setup spool join mid-print in the firmware should I forget or in some wild case, realize I need to do two joins after the print has started.

petej1940, 2024-12-22:
Another vote for Spool Join in the slicer and, perhaps, in Prusa Connect.

HAL09001, 2024-12-25:
+1 for  slicer as a default setting,
Prusa Connect would be a nice add on.

tlharris, 2024-12-31:
Another vote please for Slicer-side spool join. It's a great feature, but I'm currently sitting 2 hours away from my printer back in the shop. I have 2 spools loaded up and would love to start a print to run overnight and be ready in the morning. I'm just not sure which spool has more filament on it and if it's enough. Remote spool-join would solve this. Thanks!

agent462, 2025-01-22:
Another vote!

Waahnsinn, 2025-01-25:
+1 please Add - long time no update from Prusa...

FaboElChapo, 2025-01-26:
+1. Such a useful feature and such a shame it needs to be manually configured

Id even settle for being able to configure it on the Prusa XL Settings and it automatically applying to prints

AdamRiesel, 2025-01-31:
almost a year down the line

+1

juergen852, 2025-01-31:
There are many open issues back till  Jan 10, 2017 and I have the feeling, that many of them might have been resolved meanwhile.

I think, that old Issues like this one are "out of sight" of Prusa developers.
I have no idea, on how to attract their attention, to get at least a reply like "Won't do" or "Can't be done because....".

Does anybody of you "Spool Join Supporters" have an idea on how to get a feedback from them?

eqdeng, 2025-02-03:
Adding my support for something like this, or issue [4264](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4264), or [4432](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4432). The way Spool Join is currently implemented means I can't use the print queue when one of our spools is low. I don't particularly care whether it's in Slicer, or Connect, or available on the machine console mid-print, so long as I can set it up without cancelling queued prints.

FaboElChapo, 2025-02-05:
Im not sure this works, but it seems this actually is a feature in Prusa Connect
If you click on a print file and scroll to the bottom, you can assign filaments and then also spool join, just like on the printer
Might not be what everyone needs, but for me this totally works!

[image]

eqdeng, 2025-02-05:
That must be new! I don't believe that was available within the past few weeks; thank you for bringing it to our attention, @FaboElChapo

buchnerb, 2025-02-05:
Where in Prusa Connect are you clicking on a print file. I tried several places and can't find it.Buchnerb

 
 

grafik.png (view on web)

FaboElChapo, 2025-02-05:
> Where in Prusa Connect are you clicking on a print file. I tried several places and can't find it.Buchnerb

From the main view:
Print files (tab) > click any gcode file >scroll down to "Tool Mapping" > Hover over a gcode filament(left side) and press "assign slot" > assign filaments as you would do on the printer

eqdeng, 2025-02-05:
@buchnerb Go to the specific printer page, then the Printer Files tab:

[image]

Click on the gcode file you want to queue up. Scroll down to the bottom of the details of the printer file:

[image]

agent462, 2025-02-05:
This is awesome!  Still would be awesome to define this from Slicer.

nhustak, 2025-02-05:
OK that's great didn't know it was there.     But yeah, will be some much better when we can do it in the slicer.  I rarely even use that interface anymore - 'send & print' is my motto.

agent462, 2025-02-05:
I tried to use it and was met with this.  Running latest XL firmware.

[image]

juergen852, 2025-02-18:
As far as I know, you can define spool join also at the printer's display before starting the print.
While the print is running, you can't modify spool join manually at the printer.
Also something Prusa should add to the firmware....

As I use Repetier or Octoprint, defining spool join in P-Connect is not feasible for me.

Prusa Connect is no option for us, as it is cloud based.
Prusa say that data transfer is secure using SSL etc., but as they to not mention "stored encrypted" or "we do not access or share your data"....
Also some features are missing in PC like multiple livecams with 10-20fps or time lapse.

AndrewParry564, 2025-05-11:
This would be amazing in Slicer!!!

Jawaswag, 2025-05-17:
+1 from me as well. I use leftover filament to print my prototypes and setting the spool join function would ease my workflow tremendously. My PC and printer are literally at the opposite ends of my house.

Jawaswag, 2025-07-24:
Hi all,

This can already be done in PrusaSlicer. Select your printer in Prusa Connect. Select the print file. Scroll down to Tool Mapping. You can select the toolheads and spool join.

Thanks to the devs who made it possible!

nhustak, 2025-11-25:
Well....My XL runs like a clock.  I have issues with Prusa but the printer itself is pretty awesome.   Bambu has it's own issues.   Let's not make this an xbox/playstation thing.

devicenul, 2026-09-03:
Does legacy mean it's planned to be in 3.0.0 in some other form? I mean, this seems a pretty straightforwarad QoL issue for users and I'm shocked it hasn't been tackeled.

HespoHusky, 2026-09-09:
Please don't drop this issue

riczard-chamond, 2026-09-09:
This is actually a must have in 3.0.0, hope it will be there, hopefully sooner than later

ThatWileyGuy, 2026-10-11:
> 20\. Im not sure this works, but it seems this actually is a feature in Prusa Connect
>          If you click on a print file and scroll to the bottom, you can assign filaments and then also spool join, just like on the printer
>          Might not be what everyone needs, but for me this totally works!
>          [image]

I used to be able to use this but it's missing now. Anyone else seeing the same?